pgaskin/kobopatch-patches #104: Patches for 4.31.19086

pgaskin, 2022-01-20:
- [ ] Upload testdata (@pgaskin)
- [ ] Extract resources (@pgaskin)
- [ ] Find libnickel offsets where necessary (@pgaskin)
- [ ] Update most patches (@pgaskin)
- [ ] Update nickel patches if necessary (@jackiew1?)
- [ ] Look into selection menu changes for pgaskin/NickelMenu#119
- [ ] Test patches (@pgaskin)
- [ ] Make release (@pgaskin)
- [ ] MR post (@pgaskin)

jackiew1, 2022-01-21:
@pgaskin 

I'm planning to add the beta patch from #93, `KePub stylesheet additions - optimizeLegibility`, to the libnickel yaml file for this release, so please don't release patches until I've been able to do it. I don't see 4.31.19086 in the `versions` section yet, so can't do it at the moment.

Also it's midnight here so not much time for me to do any testing tonight. I will have time tomorrow, though.

Is there some reason why your firmware page doesn't have a link for the Mark9 fw for the Libra2?

pgaskin, 2022-01-21:
> I'm planning to add the beta patch from #93, KePub stylesheet additions - optimizeLegibility, to the libnickel yaml file for this release, so please don't release patches until I've been able to do it. I don't see 4.31.19086 in the versions section yet, so can't do it at the moment.

OK, I'll wait until tomorrow. It'll be in the versions section as soon as I take a quick look at libnickel and extract nickel's resources (probably within the next 30-60 minutes).

> Is there some reason why your firmware page doesn't have a link for the Mark9 fw for the Libra2?

It doesn't appear to have been released for the Libra2 to any of the affiliates.

pgaskin, 2022-01-21:
**Resource extraction:**

```bash
./qrc2zip --output "nickel.19086.qInitResources_resources.zip"    --recursive --verbose "nickel" 1 $((0x12de920 - 0x0010000)) $((0x00271f0 - 0x0010000)) $((0x12de428 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
./qrc2zip --output "nickel.19086.qInitResources_translations.zip" --recursive --verbose "nickel" 1 $((0x1675568 - 0x0010000)) $((0x12ded10 - 0x0010000)) $((0x16752e0 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
./qrc2zip --output "nickel.19086.qInitResources_styles.zip"       --recursive --verbose "nickel" 1 $((0x1695328 - 0x0010000)) $((0x1675680 - 0x0010000)) $((0x1692840 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
./qrc2zip --output "nickel.19086.qInitResources_certificates.zip" --recursive --verbose "nickel" 1 $((0x16bc900 - 0x0010000)) $((0x1695f30 - 0x0010000)) $((0x16bb8b0 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
```

<details><summary><b>Output</b></summary>

```
nickel@19086/qInitResources_resources
DIR     "qss" (0x12CED02 + 14)
FILE    "qss/TouchMenu.qss" (0x12CDA1A + 189)
DIR     "html" (0x12CECE6 + 28)
FILE    "html/RGBFrontLightAboutView.html" (0x12CDFAB + 1144)
FILE    "html/RGBFrontLightAboutViewSnow.html" (0x12CDADB + 1228)
DIR     "keys" (0x12CECCA + 28)
FILE    "keys/device.qmap" (0x12CD8C3 + 124)
FILE    "keys/avrcp.qmap" (0x12CD6E7 + 472)
FILE    "omfTemplate.html" (0x114AA04 + 231)
DIR     "pika.rcc" (0x410D + 14)
DIR     "pika.rcc/images" (0x411B + 42)
DIR     "pika.rcc/images/fte" (0x4161 + 14)
FILE    "pika.rcc/images/fte/quicktour_diagram.png" (0x110D + 3841)
DIR     "pika.rcc/images/home" (0x4153 + 14)
FILE    "pika.rcc/images/home/quicktour_device.png" (0x18 + 4337)
DIR     "pika.rcc/images/about" (0x4145 + 14)
FILE    "pika.rcc/images/about/global_certification.png" (0x2012 + 8221)
FILE    "verifier.pub" (0xEDB54F + 451)
DIR     "daylight.rcc" (0xD5FC2 + 14)
DIR     "daylight.rcc/images" (0xD5FD0 + 350)
DIR     "daylight.rcc/images/fte" (0xD78EA + 280)
FILE    "daylight.rcc/images/fte/missingbooks_factoryreset.png" (0xB75B8 + 9065)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg4_adjust_brightness.png" (0x77175 + 24664)
FILE    "daylight.rcc/images/fte/missingbooks_account.png" (0xACB08 + 22560)
FILE    "daylight.rcc/images/fte/missingbooks_repair.png" (0x72070 + 20737)
FILE    "daylight.rcc/images/fte/quicktour_nightmode.png" => "daylight.rcc/images/fte/quicktour_nightmode[language!Japanese].png" (0xA6ABE + 24646)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg2.png" => "daylight.rcc/images/fte/quicktour_reading_pg2[language!Japanese].png" (0x886B8 + 23892)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg2.png" (0x8E410 + 23662)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg3.png" => "daylight.rcc/images/fte/quicktour_reading_pg3[language!Japanese].png" (0x94082 + 23716)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg3.png" (0x99D2A + 23519)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg5.png" (0xA1DA3 + 19735)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg1.png" (0x82D9E + 22806)
FILE    "daylight.rcc/images/fte/missingbooks_sync.png" (0xB9925 + 6061)
FILE    "daylight.rcc/images/fte/fte_oob_connected.png" => "daylight.rcc/images/fte/fte_oob_connected[language!Japanese].png" (0x6FCA7 + 9157)
FILE    "daylight.rcc/images/fte/fte_oob_plugin.svg" (0xB680E + 3494)
FILE    "daylight.rcc/images/fte/pocket_logo_large.png" (0xBB0D6 + 3862)
FILE    "daylight.rcc/images/fte/quicktour_texticon_search.png" (0x82CFC + 158)
FILE    "daylight.rcc/images/fte/quicktour_diagram.png" (0x9F90D + 8289)
FILE    "daylight.rcc/images/fte/missingbooks_help.png" (0x7D1D1 + 23335)
FILE    "daylight.rcc/images/fte/overflow_small.png" (0xA1972 + 1069)
FILE    "daylight.rcc/images/fte/missingbooks_intro.png" (0xB232C + 17630)
DIR     "daylight.rcc/images/book" (0xD787A + 112)
FILE    "daylight.rcc/images/book/dustbin_delete_icon_inverted.png" (0x582D7 + 257)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_highlight.png" (0x584A8 + 447)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_dogear.png" (0x57FF7 + 409)
FILE    "daylight.rcc/images/book/markup_icon_transparent_padding.png" (0x58739 + 994)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_note.png" (0x5866B + 202)
FILE    "daylight.rcc/images/book/note_icon_transparent_padding.png" (0x583DC + 200)
FILE    "daylight.rcc/images/book/dustbin_delete_icon.png" (0x58194 + 319)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_markup.png" (0x58B1F + 884)
DIR     "daylight.rcc/images/help" (0xD7850 + 42)
FILE    "daylight.rcc/images/help/help_large.png" (0x6A74A + 1138)
FILE    "daylight.rcc/images/help/help_icon.png" (0x6A348 + 1022)
FILE    "daylight.rcc/images/help/help_redeem_audiobook_credit.png" (0x6ABC0 + 5145)
DIR     "daylight.rcc/images/home" (0xD772A + 294)
FILE    "daylight.rcc/images/home/main_nav_books.png" (0x4B8A0 + 637)
FILE    "daylight.rcc/images/home/my_notebooks_book.png" (0x3CAC7 + 201)
[... 2523 lines cut ...]
FILE    "dragon.rcc/images/statusbar/battery_14.png" (0xCE85 + 210)
FILE    "dragon.rcc/images/statusbar/battery_13.png" (0x4297 + 211)
FILE    "dragon.rcc/images/statusbar/battery_15.png" (0x9DC7 + 211)
FILE    "dragon.rcc/images/statusbar/battery_10.png" (0xB291 + 208)
FILE    "dragon.rcc/images/statusbar/battery_11.png" (0x8D58 + 212)
FILE    "dragon.rcc/images/statusbar/battery_16.png" (0x71C4 + 211)
[... cut ...]
FILE    "dragon.rcc/images/statusbar/battery_07.png" (0xF0D7 + 212)
FILE    "dragon.rcc/images/statusbar/battery_09.png" (0x9851 + 211)
FILE    "dragon.rcc/images/statusbar/statusbar_wifi_0.png" (0xD268 + 408)
FILE    "dragon.rcc/images/statusbar/statusbar_wifi_1.png" (0xA51D + 411)
FILE    "dragon.rcc/images/statusbar/statusbar_wifi_4.png" (0xE31F + 397)
FILE    "dragon.rcc/images/statusbar/statusbar_wifi_2.png" (0x7937 + 410)
FILE    "dragon.rcc/images/statusbar/statusbar_wifi_3.png" (0x5521 + 417)
F

pgaskin, 2022-01-21:
```
Errors:
  /home/patrick/src/kobopatch-patches/src/versions/4.31.19086/libadobe.so.yaml/pgaskin.yaml: Remove PDF map widget shown during panning: could not apply patch "Remove PDF map widget shown during panning": line 9: inst 4: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.31.19086/libnickel.so.1.0.0.yaml/pgaskin.yaml: Both page turn sides go next: could not apply patch "Both page turn sides go next": line 17: inst 4: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.31.19086/libnickel.so.1.0.0.yaml/pgaskin.yaml: Customize ComfortLight settings: could not apply patch "Customize ComfortLight settings": line 553: inst 4: ReplaceInt: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.31.19086/libnickel.so.1.0.0.yaml/pgaskin.yaml: Replace adobe page numbers toggle with invert screen: could not apply patch "Replace adobe page numbers toggle with invert screen": line 44: inst 4: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.31.19086/nickel.yaml/pgaskin.yaml: Increase size of kepub chapter progress chart: could not apply patch "Increase size of kepub chapter progress chart": line 28: inst 4: ReplaceZlib: not a zlib stream
  /home/patrick/src/kobopatch-patches/src/versions/4.31.19086/nickel.yaml/pgaskin.yaml: Remove forgot pin button from lock screen: could not apply patch "Remove forgot pin button from lock screen": line 20: inst 4: ReplaceZlib: not a zlib stream
```

The usual, but also `Both page turn sides go next` (this one will probably need new offsets) and `Replace adobe page numbers toggle with invert screen` (this one might go).

pgaskin, 2022-01-21:
@jackiew, I've pushed most of the patches.

I may need to rewrite `Customize ComfortLight settings`.

jackiew1, 2022-01-21:
@pgaskin, I've added the new patch `KePub stylesheet additions - optimizeLegibility` (_or more precisely reinstated it_) with a caveat in the Description about possible side effects when highlighting vertical text.

From your Error report it doesn't look like any of my patches need changing so I'll leave it to you to decide when to do the release, with or without `Customize ComfortLight settings`.

pgaskin, 2022-01-21:
Thanks for taking the time to get this done tonight, @jackiew1.

I will most likely do the release tonight after I deal with the NickelMenu regression and backport your patch updates from November. I will also update the ComfortLight patch for the release too as long as the logic is mostly the same.

pgaskin, 2022-01-21:
Released: [`v74`](https://github.com/pgaskin/kobopatch-patches/releases/v74), MR:thread-344610.

pgaskin, 2022-01-21:
<details><summary><b>$ git diff --no-index src/versions/{4.30.18838,4.31.19086}</b></summary>

```diff
diff --git a/src/versions/4.30.18838/libadobe.so.yaml/pgaskin.yaml b/src/versions/4.31.19086/libadobe.so.yaml/pgaskin.yaml
index 78af60a..81032e6 100644
--- a/src/versions/4.30.18838/libadobe.so.yaml/pgaskin.yaml
+++ b/src/versions/4.31.19086/libadobe.so.yaml/pgaskin.yaml
@@ -6,5 +6,5 @@ Remove PDF map widget shown during panning:
   - BaseAddress:  {Sym: "N3AdobeReader::showMapWidget()"}
   # tail: N3AdobeReader::updatePanningMap() -> N3AdobeReader::hideMapWidget()
   # TODO: figure out what broke the plt parsing in kobopatch for libadobe in 18730+
-  - ReplaceBytes: {Offset:  80, FindInstBW: 0x1BD1C, ReplaceInstBW: 0x1B08C}
-  - ReplaceBytes: {Offset: 192, FindInstBW: 0x1BD1C, ReplaceInstBW: 0x1B08C}
+  - ReplaceBytes: {Offset:  80, FindInstBW: 0x1BDE8, ReplaceInstBW: 0x1B11C}
+  - ReplaceBytes: {Offset: 192, FindInstBW: 0x1BDE8, ReplaceInstBW: 0x1B11C}
diff --git a/src/versions/4.30.18838/libnickel.so.1.0.0.yaml/jackie_w.yaml b/src/versions/4.31.19086/libnickel.so.1.0.0.yaml/jackie_w.yaml
index 2e9858d..876338b 100644
--- a/src/versions/4.30.18838/libnickel.so.1.0.0.yaml/jackie_w.yaml
+++ b/src/versions/4.31.19086/libnickel.so.1.0.0.yaml/jackie_w.yaml
@@ -121,3 +121,25 @@ KePub stylesheet additions - optimizeSpeed:
   # *** Don't change anything in the 4 ReplaceString lines above ***
   #
   - FindReplaceString: {Find: "/*********************************/", Replace: "body{text-rendering:optimizeSpeed }", MustMatchLength: yes}
+
+KePub stylesheet additions - optimizeLegibility:
+  - Enabled: no
+  - PatchGroup: KePub stylesheet additions alternatives
+  - Description: |
+        Enables ligatures/kerning in KePubs, but may cause some additional
+        rendering problems for fully justified text.
+        N.B. May have unexpected side effects when highlighting text in languages
+             which display vertical text, e.g. Chinese/Japanese
+  - FindBaseAddressString: ".KBSearchResult, .KBAnnotation, .KBHighlighting {"
+  # *** Don't change anything in the 4 ReplaceString lines below ***
+  # Create some room to add the extra CSS
+  - ReplaceString: {Find: ".KBSearchResult, .KBAnnotation, .KBHighlighting { font-size: 100% !important; -webkit-text-combine: inherit !important; }\n", Replace: ".KBSearchResult,.KBAnnotation,.KBHighlighting{font-size:100%!important;-webkit-text-combine:inherit!important}.KBAnnotatio", MustMatchLength: yes}
+  - ReplaceString: {Offset: 122, Find: ".KBAnnotation[writingMode=\"horizontal-tb\"] { border-bottom: 2px solid black !important; }\n", Replace: "n[writingMode=\"horizontal-tb\"]{border-bottom:2px solid #000!important}.KBAnnotation[writin", MustMatchLength: yes}
+  - ReplaceString: {Offset: 212, Find: ".KBAnnotation[writingMode=\"vertical-rl\"] { border-right: 2px solid black !important; }\n", Replace: "gMode=\"vertical-rl\"]{border-right:2px solid #000!important}.KBAnnotation[writingMode=\"v", MustMatchLength: yes}
+  - ReplaceString: {Offset: 299, Find: ".KBAnnotation[writingMode=\"vertical-lr\"] { border-left: 2px solid black !important; }", Replace: "ertical-lr\"]{border-left:2px solid #000}/*******************************************/", MustMatchLength: yes}
+  # *** Don't change anything in the 4 ReplaceString lines above ***
+  #
+  - FindReplaceString:
+        Find:    "/*******************************************/"
+        Replace: "body{text-rendering:optimizeLegibility      }"
+        MustMatchLength: yes
diff --git a/src/versions/4.30.18838/libnickel.so.1.0.0.yaml/pgaskin.yaml b/src/versions/4.31.19086/libnickel.so.1.0.0.yaml/pgaskin.yaml
index 6e08421..facaeff 100644
--- a/src/versions/4.30.18838/libnickel.so.1.0.0.yaml/pgaskin.yaml
+++ b/src/versions/4.31.19086/libnickel.so.1.0.0.yaml/pgaskin.yaml
@@ -13,7 +13,7 @@ Both page turn sides go next:
   - Enabled: no
   - Description: Make both page turn sides go next in the reader.
   # In ReadingView::processTap(TapGesture*) where the page direction is checked:
-  - BaseAddress:  {Sym: "ReadingView::processTap(TapGesture*)", Rel: 652}
+  - BaseAddress:  {Sym: "ReadingView::processTap(TapGesture*)", Rel: 560}
   - ReplaceBytes: {Offset:  0, FindInstBLX: {SymPLT: "ReadingView::leftToRightPageProgressDirection() const"}, CheckOnly: true}
   # nop the branch for checking if left-to-right (so it always runs the swap of back/forward)
   - ReplaceBytes: {Offset:  4, FindH: 18 B9, ReplaceInstNOP: true}
@@ -41,7 +41,7 @@ Replace adobe page numbers toggle with invert screen:
       See https://github.com/pgaskin/NickelMenu/issues/111 for more information.
   # Settings page
   - BaseAddress:  "N3SettingsReadingView::N3SettingsReadingView(QWidget*)"
-  - ReplaceBytes: {Offset: 914, FindInstBLX: {SymPLT: "ReadingSettings::getShowAdobePageNumbers()"},     ReplaceInstBLX: {SymPLT: "FeatureSettings::invertScreen()"}}
+  - ReplaceBytes: {Offset: 900, FindInstBLX: {SymPLT: "ReadingSettings::getShowAdobePageNumbers()"},     ReplaceInstBLX: {SymPLT: "FeatureSettings::invertScreen()"}}
   # Settings page controller
   - BaseAddress:  {Sym: "N3SettingsReadingController::showAdobePageNumbersToggled()"}
   - ReplaceBytes: {Offset:  74, FindInstBLX: {SymPLT: "ReadingSettings::getShowAdobePageNumbers()"},     ReplaceInstBLX: {SymPLT: "FeatureSettings::invertScreen()"}}
@@ -542,7 +542,7 @@ Customize ComfortLight settings:
   # In an unnamed subroutine two layers into FrontLightPopupController::loadView
[... cut ...]
       Replace: "#lblForgotPin,#lblSignOut{qproperty-visible:false;}"
@@ -23,7 +23,7 @@ Remove forgot pin button from lock screen:
 Increase size of kepub chapter progress chart:
   - Enabled: no
   - Description: Originally by oren64, rewritten for 4.16.13337 by pgaskin (geek1011).
-  - BaseAddress: 0x167D310 # qss/ReadingMenuStatsView.qss
+  - BaseAddress: 0x1667475 # qss/ReadingMenuStatsView.qss
   - ReplaceZlibGroup:
       Replacements:
       # Top padding is already set to 15px, 25px, 33px, vertical aligned to middle.
```

</details>

closed 2022-01-21 by pgaskin