pgaskin/kobopatch-patches #105: Patches for 4.32.19501

pgaskin, 2022-04-14:
- [ ] Upload testdata (@pgaskin)
- [ ] Extract resources (@pgaskin)
- [ ] Find libnickel offsets where necessary (@pgaskin)
- [ ] Update most patches (@pgaskin, @shermp)
- [ ] Update nickel patches if necessary (@jackiew1?)
- [ ] Look at new stuff
   - [ ] wtf is strickel
- [ ] Test patches (@pgaskin, @shermp, @jackiew1)
- [ ] Make release (@pgaskin)
- [ ] MR post (@pgaskin)

pgaskin, 2022-04-14:
**Resource extraction:**

```bash
./qrc2zip --output "nickel.19086.qInitResources_resources.zip"    --recursive --verbose "nickel" 1 $((0x12de920 - 0x0010000)) $((0x00271f0 - 0x0010000)) $((0x12de428 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
./qrc2zip --output "nickel.19086.qInitResources_translations.zip" --recursive --verbose "nickel" 1 $((0x1675568 - 0x0010000)) $((0x12ded10 - 0x0010000)) $((0x16752e0 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
./qrc2zip --output "nickel.19086.qInitResources_styles.zip"       --recursive --verbose "nickel" 1 $((0x1695328 - 0x0010000)) $((0x1675680 - 0x0010000)) $((0x1692840 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
./qrc2zip --output "nickel.19086.qInitResources_certificates.zip" --recursive --verbose "nickel" 1 $((0x16bc900 - 0x0010000)) $((0x1695f30 - 0x0010000)) $((0x16bb8b0 - 0x0010000)) || { echo "Error: qrc2zip failed." 1>&2; exit 1; }
```

<details><summary><b>Output</b></summary>

```
nickel@19086/qInitResources_resources
DIR     "qss" (0x12CED02 + 14)
FILE    "qss/TouchMenu.qss" (0x12CDA1A + 189)
DIR     "html" (0x12CECE6 + 28)
FILE    "html/RGBFrontLightAboutView.html" (0x12CDFAB + 1144)
FILE    "html/RGBFrontLightAboutViewSnow.html" (0x12CDADB + 1228)
DIR     "keys" (0x12CECCA + 28)
FILE    "keys/device.qmap" (0x12CD8C3 + 124)
FILE    "keys/avrcp.qmap" (0x12CD6E7 + 472)
FILE    "omfTemplate.html" (0x114AA04 + 231)
DIR     "pika.rcc" (0x410D + 14)
DIR     "pika.rcc/images" (0x411B + 42)
DIR     "pika.rcc/images/fte" (0x4161 + 14)
FILE    "pika.rcc/images/fte/quicktour_diagram.png" (0x110D + 3841)
DIR     "pika.rcc/images/home" (0x4153 + 14)
FILE    "pika.rcc/images/home/quicktour_device.png" (0x18 + 4337)
DIR     "pika.rcc/images/about" (0x4145 + 14)
FILE    "pika.rcc/images/about/global_certification.png" (0x2012 + 8221)
FILE    "verifier.pub" (0xEDB54F + 451)
DIR     "daylight.rcc" (0xD5FC2 + 14)
DIR     "daylight.rcc/images" (0xD5FD0 + 350)
DIR     "daylight.rcc/images/fte" (0xD78EA + 280)
FILE    "daylight.rcc/images/fte/missingbooks_factoryreset.png" (0xB75B8 + 9065)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg4_adjust_brightness.png" (0x77175 + 24664)
FILE    "daylight.rcc/images/fte/missingbooks_account.png" (0xACB08 + 22560)
FILE    "daylight.rcc/images/fte/missingbooks_repair.png" (0x72070 + 20737)
FILE    "daylight.rcc/images/fte/quicktour_nightmode.png" => "daylight.rcc/images/fte/quicktour_nightmode[language!Japanese].png" (0xA6ABE + 24646)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg2.png" => "daylight.rcc/images/fte/quicktour_reading_pg2[language!Japanese].png" (0x886B8 + 23892)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg2.png" (0x8E410 + 23662)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg3.png" => "daylight.rcc/images/fte/quicktour_reading_pg3[language!Japanese].png" (0x94082 + 23716)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg3.png" (0x99D2A + 23519)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg5.png" (0xA1DA3 + 19735)
FILE    "daylight.rcc/images/fte/quicktour_reading_pg1.png" (0x82D9E + 22806)
FILE    "daylight.rcc/images/fte/missingbooks_sync.png" (0xB9925 + 6061)
FILE    "daylight.rcc/images/fte/fte_oob_connected.png" => "daylight.rcc/images/fte/fte_oob_connected[language!Japanese].png" (0x6FCA7 + 9157)
FILE    "daylight.rcc/images/fte/fte_oob_plugin.svg" (0xB680E + 3494)
FILE    "daylight.rcc/images/fte/pocket_logo_large.png" (0xBB0D6 + 3862)
FILE    "daylight.rcc/images/fte/quicktour_texticon_search.png" (0x82CFC + 158)
FILE    "daylight.rcc/images/fte/quicktour_diagram.png" (0x9F90D + 8289)
FILE    "daylight.rcc/images/fte/missingbooks_help.png" (0x7D1D1 + 23335)
FILE    "daylight.rcc/images/fte/overflow_small.png" (0xA1972 + 1069)
FILE    "daylight.rcc/images/fte/missingbooks_intro.png" (0xB232C + 17630)
DIR     "daylight.rcc/images/book" (0xD787A + 112)
FILE    "daylight.rcc/images/book/dustbin_delete_icon_inverted.png" (0x582D7 + 257)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_highlight.png" (0x584A8 + 447)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_dogear.png" (0x57FF7 + 409)
FILE    "daylight.rcc/images/book/markup_icon_transparent_padding.png" (0x58739 + 994)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_note.png" (0x5866B + 202)
FILE    "daylight.rcc/images/book/note_icon_transparent_padding.png" (0x583DC + 200)
FILE    "daylight.rcc/images/book/dustbin_delete_icon.png" (0x58194 + 319)
FILE    "daylight.rcc/images/book/AnnotationsList_icon_markup.png" (0x58B1F + 884)
DIR     "daylight.rcc/images/help" (0xD7850 + 42)
FILE    "daylight.rcc/images/help/help_large.png" (0x6A74A + 1138)
FILE    "daylight.rcc/images/help/help_icon.png" (0x6A348 + 1022)
FILE    "daylight.rcc/images/help/help_redeem_audiobook_credit.png" (0x6ABC0 + 5145)
DIR     "daylight.rcc/images/home" (0xD772A + 294)
FILE    "daylight.rcc/images/home/main_nav_books.png" (0x4B8A0 + 637)
FILE    "daylight.rcc/images/home/my_notebooks_book.png" (0x3CAC7 + 201)
[... 2523 lines cut ...]
FILE    "dragon.rcc/images/statusbar/battery_14.png" (0xCE85 + 210)
FILE    "dragon.rcc/images/statusbar/battery_13.png" (0x4297 + 211)
FILE    "dragon.rcc/images/statusbar/battery_15.png" (0x9DC7 + 211)
FILE    "dragon.rcc/images/statusbar/battery_10.png" (0xB291 + 208)
FILE    "dragon.rcc/images/statusbar/battery_11.png" (0x8D58 + 212)
FILE    "dragon.rcc/images/statusbar/battery_16.png" (0x71C4 + 211)
[... cut ...]
FILE    "dragon.rcc/images/statusbar/battery_07.png" (0xF0D7 + 212)
FILE    "dragon.rcc/images/statusbar/battery_09.png" (0x9851 + 211)
FILE    "dragon.rcc/images/statusbar/statusbar_wifi_0.png" (0xD268 + 408)
FILE    "dragon.rcc/images/statusbar/statusbar_wifi_1.png" (0xA51D + 411)
FILE    "dragon.rcc/images/statusbar/statusbar_wifi_4.png" (0xE31F + 397)
FILE    "dragon.rcc/images/statusbar/statusbar_wifi_2.png" (0x7937 + 410)
FILE    "dragon.rcc/images/statusbar/statusbar_wifi_3.png" (0x5521 + 417)
F

shermp, 2022-04-14:
>  wtf is strickel
>  and fontickle

Gotta love their binary naming...

pgaskin, 2022-04-14:
```
Errors:
  /home/patrick/src/kobopatch-patches/src/versions/4.32.19501/libadobe.so.yaml/pgaskin.yaml: Remove PDF map widget shown during panning: could not apply patch "Remove PDF map widget shown during panning": line 9: inst 4: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.32.19501/libnickel.so.1.0.0.yaml/geoffr.yaml: KePub constant font sharpness: could not apply patch "KePub constant font sharpness": line 409: inst 3: FindReplaceString: FindBaseAddressString: FindBaseAddress: could not find bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.32.19501/libnickel.so.1.0.0.yaml/geoffr.yaml: Custom font sizes: could not apply patch "Custom font sizes": line 197: inst 4: ReplaceInt: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.32.19501/libnickel.so.1.0.0.yaml/geoffr.yaml: Ignore .otf fonts: could not apply patch "Ignore .otf fonts": line 486: inst 3: FindReplaceString: FindBaseAddressString: FindBaseAddress: could not find bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.32.19501/libnickel.so.1.0.0.yaml/geoffr.yaml: My 24 line spacing values: could not apply patch "My 24 line spacing values": line 51: inst 6: ReplaceBytes: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.32.19501/libnickel.so.1.0.0.yaml/geoffr.yaml: Set KePub hyphenation: could not apply patch "Set KePub hyphenation": line 326: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.32.19501/librmsdk.so.1.0.0.yaml/geoffr.yaml: Default ePub serif font (Amasis): could not apply patch "Default ePub serif font (Amasis)": line 26: inst 3: FindReplaceString: FindBaseAddressString: FindBaseAddress: could not find bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.32.19501/libnickel.so.1.0.0.yaml/sherman.yaml: Unify font sizes: could not apply patch "Unify font sizes": line 27: inst 4: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.32.19501/libnickel.so.1.0.0.yaml/pgaskin.yaml: Replace adobe page numbers toggle with invert screen: could not apply patch "Replace adobe page numbers toggle with invert screen": line 44: inst 4: ReplaceBytes: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.32.19501/libnickel.so.1.0.0.yaml/pgaskin.yaml: Customize ComfortLight settings: could not apply patch "Customize ComfortLight settings": line 549: inst 4: ReplaceInt: could not find specified bytes at offset
  /home/patrick/src/kobopatch-patches/src/versions/4.32.19501/libnickel.so.1.0.0.yaml/pgaskin.yaml: Enable advanced settings for all fonts: could not apply patch "Enable advanced settings for all fonts": line 520: inst 3: ReplaceBytes: expand FindInstBLX=kobopatch.FlexAbsOffset{Offset:(*int32)(nil), Sym:(*string)(nil), SymPLT:(*string)(0xc004494530), SymPLTTail:(*string)(nil), Inline:false, Rel:(*int32)(nil)}: ResolveSymPLT("N3FontTypeUtil::setupMonotypeDropdown(TouchDropDown*, QString const&)"): no such symbol "N3FontTypeUtil::setupMonotypeDropdown(TouchDropDown*, QString const&)"
  /home/patrick/src/kobopatch-patches/src/versions/4.32.19501/libnickel.so.1.0.0.yaml/pgaskin.yaml: Change Browse Kobo home screen link target - Activity: could not apply patch "Change Browse Kobo home screen link target - Activity": line 200: inst 4: ReplaceBytes: could not find specified bytes
  /home/patrick/src/kobopatch-patches/src/versions/4.32.19501/nickel.yaml/pgaskin.yaml: Increase size of kepub chapter progress chart: could not apply patch "Increase size of kepub chapter progress chart": line 28: inst 4: ReplaceZlib: not a zlib stream
  /home/patrick/src/kobopatch-patches/src/versions/4.32.19501/nickel.yaml/pgaskin.yaml: Remove forgot pin button from lock screen: could not apply patch "Remove forgot pin button from lock screen": line 20: inst 4: ReplaceZlib: not a zlib stream
```

Mostly the same as usual, but all the big font-related patches are now broken too.

shermp, 2022-04-14:
With a binary name like `fontickle`, and the release notes Kobo provided, I'd say they've done some font handling reworking.

pgaskin, 2022-04-14:
<details><summary>symdump</summary>

```diff
diff --git a/19086.tmp b/19501.tmp
index 0536ad1..c4ae4f3 100644
--- a/19086.tmp
+++ b/19501.tmp
@@ -1126,6 +1126,7 @@ AbstractNickelMenuController::createMenuImageItem(QMenu*, QString const&) _ZN28A
 AbstractNickelMenuController::createMenuSectionLabel(QMenu*, QString const&) _ZN28AbstractNickelMenuController22createMenuSectionLabelEP5QMenuRK7QString
 AbstractNickelMenuController::createMenuTextItem(QMenu*, QString const&, bool, bool, QString const&) _ZN28AbstractNickelMenuController18createMenuTextItemEP5QMenuRK7QStringbbS4_
 AbstractNickelMenuController::createMenuTextItemWithExtraField(QMenu*, QString const&, QString const&, bool, bool, QString const&) _ZN28AbstractNickelMenuController32createMenuTextItemWithExtraFieldEP5QMenuRK7QStringS4_bbS4_
+AbstractNickelMenuController::createMultiSelectMenuTextItem(QMenu*, QString const&, bool, QString const&) _ZN28AbstractNickelMenuController29createMultiSelectMenuTextItemEP5QMenuRK7QStringbS4_
 AbstractNickelMenuController::metaObject() const _ZNK28AbstractNickelMenuController10metaObjectEv
 AbstractNickelMenuController::onDesktopResized() _ZN28AbstractNickelMenuController16onDesktopResizedEv
 AbstractNickelMenuController::popupFromWidget() const _ZNK28AbstractNickelMenuController15popupFromWidgetEv
@@ -2069,6 +2070,11 @@ AdobeInitializer::~AdobeInitializer() _ZN16AdobeInitializerD0Ev
 AdobeInitializer::~AdobeInitializer() _ZN16AdobeInitializerD1Ev
 AdobeInitializer::~AdobeInitializer() _ZN16AdobeInitializerD2Ev
 AdobeInitializer::initializeAdobe() _ZN16AdobeInitializer15initializeAdobeEv
+AdobeMapReleaserCallback::~AdobeMapReleaserCallback() _ZN24AdobeMapReleaserCallbackD0Ev
+AdobeMapReleaserCallback::~AdobeMapReleaserCallback() _ZN24AdobeMapReleaserCallbackD1Ev
+AdobeMapReleaserCallback::~AdobeMapReleaserCallback() _ZN24AdobeMapReleaserCallbackD2Ev
+AdobeMapReleaserCallback::invoke(dp::Unknown*) _ZN24AdobeMapReleaserCallback6invokeEPN2dp7UnknownE
+AdobeMapReleaserCallback::reportError(dp::String const&) _ZN24AdobeMapReleaserCallback11reportErrorERKN2dp6StringE
 AdobeNetworkProvider::AdobeNetworkProvider() _ZN20AdobeNetworkProviderC1Ev
 AdobeNetworkProvider::AdobeNetworkProvider() _ZN20AdobeNetworkProviderC2Ev
 AdobeNetworkProvider::~AdobeNetworkProvider() _ZN20AdobeNetworkProviderD0Ev
@@ -2095,7 +2101,6 @@ AdobeResourceProvider::~AdobeResourceProvider() _ZN21AdobeResourceProviderD0Ev
 AdobeResourceProvider::~AdobeResourceProvider() _ZN21AdobeResourceProviderD1Ev
 AdobeResourceProvider::~AdobeResourceProvider() _ZN21AdobeResourceProviderD2Ev
 AdobeResourceProvider::bytesReady(unsigned int, dp::Data const&, bool) _ZN21AdobeResourceProvider10bytesReadyEjRKN2dp4DataEb
-AdobeResourceProvider::clearFontCache() _ZN21AdobeResourceProvider14clearFontCacheEv
 AdobeResourceProvider::getResourceStream(dp::String const&, unsigned int) _ZN21AdobeResourceProvider17getResourceStreamERKN2dp6StringEj
 AdobeResourceProvider::propertiesReady() _ZN21AdobeResourceProvider15propertiesReadyEv
 AdobeResourceProvider::propertyReady(dp::String const&, dp::String const&) _ZN21AdobeResourceProvider13propertyReadyERKN2dp6StringES3_
@@ -3129,6 +3134,7 @@ ArticleCoverWidget::qt_metacall(QMetaObject::Call, int, void**) _ZN18ArticleCove
 ArticleCoverWidget::qt_metacast(char const*) _ZN18ArticleCoverWidget11qt_metacastEPKc
 ArticleCoverWidget::resizeEvent(QResizeEvent*) _ZN18ArticleCoverWidget11resizeEventEP12QResizeEvent
 ArticleCoverWidget::setContent(Volume const&) _ZN18ArticleCoverWidget10setContentERK6Volume
+ArticleCoverWidget::setLabelLocales(QLocale const&) _ZN18ArticleCoverWidget15setLabelLocalesERK7QLocale
 ArticleCoverWidget::staticMetaObject _ZN18ArticleCoverWidget16staticMetaObjectE
 ArticleFilter<Volume>::~ArticleFilter() _ZN13ArticleFilterI6VolumeED0Ev
 ArticleFilter<Volume>::~ArticleFilter() _ZN13ArticleFilterI6VolumeED1Ev
@@ -3316,7 +3322,6 @@ ArticlesViewMenuController::staticMetaObject _ZN26ArticlesViewMenuController16st
 AscendingKeySorter<Volume, ExpirationKey<Volume>, Collator>::~AscendingKeySorter() _ZN18AscendingKeySorterI6Volume13ExpirationKeyIS0_E8CollatorED0Ev
 AscendingKeySorter<Volume, ExpirationKey<Volume>, Collator>::~AscendingKeySorter() _ZN18AscendingKeySorterI6Volume13ExpirationKeyIS0_E8CollatorED1Ev
 AscendingKeySorter<Volume, ExpirationKey<Volume>, Collator>::~AscendingKeySorter() _ZN18AscendingKeySorterI6Volume13ExpirationKeyIS0_E8CollatorED2Ev
-AssetCounter::setResult(QVariant const&) _ZN12AssetCounter9setResultERK8QVariant
 AsyncCoverImageWorker::AsyncCoverImageWorker() _ZN21AsyncCoverImageWorkerC1Ev
 AsyncCoverImageWorker::AsyncCoverImageWorker() _ZN21AsyncCoverImageWorkerC2Ev
 AsyncCoverImageWorker::~AsyncCoverImageWorker() _ZN21AsyncCoverImageWorkerD0Ev
@@ -3383,6 +3388,8 @@ AsyncWaiter::staticMetaObject _ZN11AsyncWaiter16staticMetaObjectE
  ATTRIBUTE_APPLICABLE_SUBSCRIPTIONS
  ATTRIBUTE_ATTRIBUTION
  ATTRIBUTE_ATTRIBUTION_KANA
+ ATTRIBUTE_AUDIOBOOK_DURATION
+ ATTRIBUTE_AUDIOBOOK_IS_ABRIDGED
  ATTRIBUTE_AUTHOR_DISPLAY_NAME
  ATTRIBUTE_AVERAGE_RATING
  ATTRIBUTE_BOOK_ID
@@ -3517,6 +3524,10 @@ AsyncWaiter::staticMetaObject _ZN11AsyncWaiter16staticMetaObjectE
  ATTRIBUTE_SHELF_NAME
  ATTRIBUTE_SHELF_SYNC_TIME
  ATTRIBUTE_SHELF_TYPE
+ ATTRIBUTE_STORE_PAGES
+ ATTRIBUTE_STORE_TIME_TO_READ_LOWER_ESTIMATE
+ ATTRIBUTE_STORE_TIME_TO_READ_UPPER_ESTIMATE
+ ATTRIBUTE_STORE_WORDS
  ATTRIBUTE_SUB_ID
  ATTRIBUTE_SUBTITLE
[... 1738 lines cut ...]
 SyncMenuController::loadView() _ZN18SyncMenuController8loadViewEv
[... cut ...]
 TabManager::getLibraryFeaturedListsTab() _ZN10TabManager26getLibraryFeaturedListsTabEv
 TabManager::isDailyDealPlaceholder(Tab const&) _ZN10TabManager22isDailyDealPlaceholderERK3Tab
 TabManager::isNoResultsTab(Tab const&) _ZN10TabManager14isNoResultsTabERK3Tab
+TabManager::isStorefrontTab(Tab const&) _ZN10TabManager15isStorefrontTabERK3Tab
 TabManager::isSubsPromoPlaceholder(Tab const&) _ZN10TabManager22isSubsPromoPlaceholderERK3Tab
 TabManager::save(Tab const&, QString const&) _ZN10TabManager4saveERK3TabRK7QString
 TabManager::save(Tab con

pgaskin, 2022-04-14:
<details><summary>fontickle</summary>

```
 abort
 __aeabi_atexit
 __aeabi_uidivmod
 __aeabi_unwind_cpp_pr0
 __aeabi_unwind_cpp_pr1
bool QtMetaTypePrivate::QAssociativeIterableImpl::equalIterImpl<QMap<QString, QString> >(void* const*, void* const*) _ZN17QtMetaTypePrivate24QAssociativeIterableImpl13equalIterImplI4QMapI7QStringS3_EEEbPKPvS7_
bool QtMetaTypePrivate::QSequentialIterableImpl::equalIterImpl<QList<QMap<QString, QVariant> > >(void* const*, void* const*) _ZN17QtMetaTypePrivate23QSequentialIterableImpl13equalIterImplI5QListI4QMapI7QString8QVariantEEEEbPKPvSA_
 __bss_end__
 _bss_end__
 __bss_start
 __bss_start__
 __cxa_begin_catch
 __cxa_end_catch
 __cxa_end_cleanup
 __cxa_guard_abort
 __cxa_guard_acquire
 __cxa_guard_release
 __cxa_rethrow
 __data_start
 data_start
 _edata
 __end__
 _end
 _fini
FontickelAdaptor::bestFontForLanguage(QString const&) _ZN16FontickelAdaptor19bestFontForLanguageERK7QString
FontickelAdaptor::busy() _ZN16FontickelAdaptor4busyEv
FontickelAdaptor::findFont(int) _ZN16FontickelAdaptor8findFontEi
FontickelAdaptor::FontickelAdaptor(QObject*) _ZN16FontickelAdaptorC1EP7QObject
FontickelAdaptor::FontickelAdaptor(QObject*) _ZN16FontickelAdaptorC2EP7QObject
FontickelAdaptor::~FontickelAdaptor() _ZN16FontickelAdaptorD0Ev
FontickelAdaptor::~FontickelAdaptor() _ZN16FontickelAdaptorD1Ev
FontickelAdaptor::~FontickelAdaptor() _ZN16FontickelAdaptorD2Ev
FontickelAdaptor::fonts() _ZN16FontickelAdaptor5fontsEv
FontickelAdaptor::metaObject() const _ZNK16FontickelAdaptor10metaObjectEv
FontickelAdaptor::pathsForFont(QString const&) _ZN16FontickelAdaptor12pathsForFontERK7QString
FontickelAdaptor::qt_metacall(QMetaObject::Call, int, void**) _ZN16FontickelAdaptor11qt_metacallEN11QMetaObject4CallEiPPv
FontickelAdaptor::qt_metacast(char const*) _ZN16FontickelAdaptor11qt_metacastEPKc
FontickelAdaptor::ready() _ZN16FontickelAdaptor5readyEv
FontickelAdaptor::shutdown() _ZN16FontickelAdaptor8shutdownEv
FontickelAdaptor::start() _ZN16FontickelAdaptor5startEv
FontickelAdaptor::staticMetaObject _ZN16FontickelAdaptor16staticMetaObjectE
FontickelService::bestFontForLanguage(QString const&) _ZN16FontickelService19bestFontForLanguageERK7QString
FontickelService::busy() _ZN16FontickelService4busyEv
FontickelService::findFont(int) _ZN16FontickelService8findFontEi
FontickelService::findFont(QFontDatabase::WritingSystem) _ZN16FontickelService8findFontEN13QFontDatabase13WritingSystemE
FontickelService::FontickelService(QObject*) _ZN16FontickelServiceC1EP7QObject
FontickelService::FontickelService(QObject*) _ZN16FontickelServiceC2EP7QObject
FontickelService::~FontickelService() _ZN16FontickelServiceD0Ev
FontickelService::~FontickelService() _ZN16FontickelServiceD1Ev
FontickelService::~FontickelService() _ZN16FontickelServiceD2Ev
FontickelService::fonts() _ZN16FontickelService5fontsEv
FontickelService::metaObject() const _ZNK16FontickelService10metaObjectEv
FontickelService::pathsForFont(QString const&) _ZN16FontickelService12pathsForFontERK7QString
FontickelService::qt_metacall(QMetaObject::Call, int, void**) _ZN16FontickelService11qt_metacallEN11QMetaObject4CallEiPPv
FontickelService::qt_metacast(char const*) _ZN16FontickelService11qt_metacastEPKc
FontickelService::ready() _ZN16FontickelService5readyEv
FontickelService::shutdown() _ZN16FontickelService8shutdownEv
FontickelService::start() _ZN16FontickelService5startEv
FontickelService::staticMetaObject _ZN16FontickelService16staticMetaObjectE
FontickelService::writingSystemForLanguage(QByteArray const&) _ZN16FontickelService24writingSystemForLanguageERK10QByteArray
FontickelWorker::abort() _ZN15FontickelWorker5abortEv
FontickelWorker::complete() _ZN15FontickelWorker8completeEv
FontickelWorker::done(QMap<QString, QString>, QMap<QString, QList<QFontDatabase::WritingSystem> >) _ZN15FontickelWorker4doneE4QMapI7QStringS1_ES0_IS1_5QListIN13QFontDatabase13WritingSystemEEE
FontickelWorker::fetchFonts() _ZN15FontickelWorker10fetchFontsEv
FontickelWorker::FontickelWorker() _ZN15FontickelWorkerC1Ev
FontickelWorker::FontickelWorker() _ZN15FontickelWorkerC2Ev
FontickelWorker::~FontickelWorker() _ZN15FontickelWorkerD0Ev
[... 985 lines cut ...]
tolower@GLIBC_2.4
TopCompressedProb(char const*, int)
TrigramValue(unsigned char const*)
typeinfo for QDBusAbstractAdaptor
typeinfo for QObject
typeinfo for std::bad_alloc@GLIBCXX_3.4
typeinfo for StrickelAdaptor
typeinfo for StrickelService
typeinfo name for StrickelAdaptor
typeinfo name for StrickelService
ui()
UTF1632BoostWhack(DetectEncodingState*, int, unsigned char)
UTF16MakeEven(DetectEncodingState*, int)
UTF7BoostWhack(DetectEncodingState*, int, unsigned char)
UTF88Sub(char, char)
void std::vector<CLD2::ResultChunk, std::allocator<CLD2::ResultChunk> >::_M_emplace_back_aux<CLD2::ResultChunk const&>(CLD2::ResultChunk const&)
void std::vector<CLD2::ResultChunk, std::allocator<CLD2::ResultChunk> >::_M_emplace_back_aux<CLD2::ResultChunk const&>(CLD2::ResultChunk const&)
vtable for __cxxabiv1::__si_class_type_info@@CXXABI_1.3
vtable for StrickelAdaptor
vtable for StrickelService
```

(this is c++filt and nm since my symdump tool seems to break on it)

</details>

shermp, 2022-04-14:
Seems I'm not the only one to think that d-bus can be useful...

Interesting to see Kobo use separate binaries for the new font management/locale stuff though, would have thought they would be part of libnickel, or perhaps plugins.

pgaskin, 2022-04-14:
Random notes:
- We're probably going to lose `KePub constant font sharpness` since there isn't enough room to replace the number while keeping the arg. I might be able to patch the `QString::arg` call to remove it, but I don't have time to test that right now.
- TODO

shermp, 2022-04-14:
`Unify font sizes` looks like it might just be an offset change.

jackiew1, 2022-04-14:
@pgaskin, Just to let you know that if there is anything for me update I'll have some time to look at it tomorrow.

pgaskin, 2022-04-14:
@jackiew1, it doesn't seem like there will be anything, but we may end up releasing tomorrow anyways since I'm somewhat busy tonight and there's quite a bit to test.

pgaskin, 2022-04-15:
WIP; @shermp tested it, and it crashes at the 15th value (2/3 of the way on the slider), which makes sense due to the alignment changes there. I'll fix it when I'm with my Kobo later tonight.

```yaml
My 24 line spacing values:
  - Enabled: no
  - PatchGroup: Line spacing values alternatives
  - Description: |
      This patch changes the values on the line spacing adjustment slider,
      increasing the number of spacing options from 15 to 24, and allowing
      much narrower spacing values to be set.
    # basically rewrite the entire code starting where values are added:
    # save r5, initialise loop counter
  - BaseAddress:  {Sym: "ReadingSettings::lineHeightScalars() const", Rel: 84} # starting at first QList<double>::append(double const&)
  - ReplaceBytes: {Offset:  0, FindInstBLX: {SymPLT: "QList<double>::append(double const&)"}, ReplaceH: A9 46 00 25}              # mov sb, r5       mov r5, #0
  - ReplaceBytes: {Offset:  4, FindH: 6C A3, ReplaceH: ED 00}                         # lsls r5, r5, #3
    # loop to load 14 vals from new table
  - ReplaceBytes: {Offset:  6, FindH: D3 E9 00 23 07 F1, ReplaceH: 20 46 04 A1 29 44} # mov r0, r4   adr r1, #0x10  add r1, r5
  - ReplaceBytes: {Offset: 12, FindH: 80 01 20 46,       ReplaceInstBLX: {SymPLT: "QList<double>::append(double const&)"}}
  - ReplaceBytes: {Offset: 16, FindH: 61 E9 1C 23,       ReplaceH: 08 35 70 2D}       # add r5, #8   cmp r5, 112 ## NOTE: 14*8=112
  - ReplaceBytes: {Offset: 20, FindInstBLX: {SymPLT: "QList<double>::append(double const&)"}, ReplaceH: 00 00 00 00}       # clear out the BLX in the way of the next two replacements
  - ReplaceBytes: {Offset: 20, FindH: 00 00,             ReplaceH: F7 D1}             # bne #-10
    # jump over new table
  - ReplaceBytes: {Offset: 22, FindH: 00 00 69 A3 D3 E9, ReplaceH: 7C E0 00 BF 00 BF} # b #0xFC
    # new table: 14 8-aligned doubles
  - BaseAddress:  {Sym: "ReadingSettings::lineHeightScalars() const", Rel: 112} # right after previous instruction
  - ReplaceBytes: {Offset:  12, FindInstBLX: {SymPLT: "QList<double>::append(double const&)"}, ReplaceH: 00 00 00 00} # first, clear out the BLXs in the way
  - ReplaceBytes: {Offset:  32, FindInstBLX: {SymPLT: "QList<double>::append(double const&)"}, ReplaceH: 00 00 00 00}
  - ReplaceBytes: {Offset:  58, FindInstBLX: {SymPLT: "QList<double>::append(double const&)"}, ReplaceH: 00 00 00 00}
  - ReplaceBytes: {Offset:  78, FindInstBLX: {SymPLT: "QList<double>::append(double const&)"}, ReplaceH: 00 00 00 00}
  - ReplaceBytes: {Offset: 100, FindInstBLX: {SymPLT: "QList<double>::append(double const&)"}, ReplaceH: 00 00 00 00}
  - ReplaceBytes: {Offset:   0, FindH: 00 23 07 F1 80 01 20 46, ReplaceH: 00 00 00 00 00 00 F0 3F} # table of 1.0f values, to be replaced later
  - ReplaceBytes: {Offset:   8, FindH: 61 E9 1A 23 00 00 00 00, ReplaceH: 00 00 00 00 00 00 F0 3F}
  - ReplaceBytes: {Offset:  16, FindH: 66 A3 D3 E9 00 23 07 F1, ReplaceH: 00 00 00 00 00 00 F0 3F}
  - ReplaceBytes: {Offset:  24, FindH: 80 01 20 46 61 E9 18 23, ReplaceH: 00 00 00 00 00 00 F0 3F}
  - ReplaceBytes: {Offset:  32, FindH: 00 00 00 00 43 F2 33 33, ReplaceH: 00 00 00 00 00 00 F0 3F}
  - ReplaceBytes: {Offset:  40, FindH: 4F F0 33 32 C3 F6 F3 73, ReplaceH: 00 00 00 00 00 00 F0 3F}
  - ReplaceBytes: {Offset:  48, FindH: 07 F1 80 01 20 46 61 E9, ReplaceH: 00 00 00 00 00 00 F0 3F}
  - ReplaceBytes: {Offset:  56, FindH: 16 23 00 00 00 00 5D A3, ReplaceH: 00 00 00 00 00 00 F0 3F}
  - ReplaceBytes: {Offset:  64, FindH: D3 E9 00 23 07 F1 80 01, ReplaceH: 00 00 00 00 00 00 F0 3F}
  - ReplaceBytes: {Offset:  72, FindH: 20 46 61 E9 14 23 00 00, ReplaceH: 00 00 00 00 00 00 F0 3F}
  - ReplaceBytes: {Offset:  80, FindH: 00 00 00 23 00 22 C3 F6, ReplaceH: 00 00 00 00 00 00 F0 3F}
  - ReplaceBytes: {Offset:  88, FindH: F8 73 07 F1 80 01 20 46, ReplaceH: 00 00 00 00 00 00 F0 3F}
  - ReplaceBytes: {Offset:  96, FindH: 61 E9 12 23 00 00 00 00, ReplaceH: 00 00 00 00 00 00 F0 3F}
  - ReplaceBytes: {Offset: 104, FindH: 54 A3 D3 E9 00 23 07 F1, ReplaceH: 00 00 00 00 00 00 F0 3F}
    # jump target after table
  - BaseAddress:  {Sym: "ReadingSettings::lineHeightScalars() const", Rel: 356} # starting at first QList<double>::append(double const&)
    # initialise loop counter
  - ReplaceBytes: {Offset:  0, FindH: 00 23 00 22,       ReplaceH: 00 BF 00 25}       # nop   movs r5, #0
    # loop to load 10 vals from old table
  - ReplaceBytes: {Offset:  4, FindH: C4 F2 08 03 07 F1, ReplaceH: 20 46 29 A1 29 44} # mov r0, r4   adr r1, #0xA4   add r1, r5 ## NOTE: adr r1, #0xA4 sets r1 to the beginning of the old table
  - ReplaceBytes: {Offset: 10, FindH: 80 01 20 46,       ReplaceInstBLX: {SymPLT: "QList<double>::append(double const&)"}}
  - ReplaceBytes: {Offset: 18, FindInstBLX: {SymPLT: "QList<double>::append(double const&)"}, ReplaceH: 00 00 00 00}       # clear out the BLX in the way of the next two replacements
  - ReplaceBytes: {Offset: 14, FindH: 61 E9 02 23 00 00, ReplaceH: 08 35 50 2D F7 D1} # adds r5, #8   cmp r5, #80   bne #-10 ## NOTE: 8*10=80
    # restore r5
  - ReplaceBytes: {Offset: 20, FindH: 00 00,             ReplaceH: 4D 46}             # mov r5, sb
    # finished replacements
  - BaseAddress: 0
    #
    # Replacement value: the number of initial line spacing values to exclude from
    # the line spacing slider; minimum 0, maximum 13. E.g. 9 will result in a
    # slider with 15 positions, starting from the 10th replace_float value below.
[... cut ...]
    # device you might find that increasing the slider one step at a time doesn't
    # always result in a line spacing increase at each step. In that case, you
    # might want to use replacement values that are not quite so close together.
    #
    # For example: (I use these values on my 758x1024 Glo)
    #
    #  0.7 0.725 0.75 0.775 0.8 0.825 0.85 0.875 0.9 0.925 0.95 0.975 1.0 1.025 ...
    #
    # or: (these might better suit the 600x800 Touch)
    #
    #  0.6 0.633 0.667 0.7 0.733 0.767 0.8 0.833 0.867 9.0 9.33 9.67 1.0 1.025 ...
```

jackiew1, 2022-04-15:
Re: libnickel patch `Dictionary text font-family/font-size/line-height`. 

This is quite weird. The patch runs without error and does what's expected on my test KA1, i.e. displays Dictionary pop-ups using Arial font rather than default Georgia.

However if I apply the exact same patched `KoboRoot.tgz` to my Forma then the pop-ups don't use Arial.

I'll investigate further but I'm lacking bright ideas. It may take some time.

pgaskin, 2022-04-15:
@jackiew1, I'm planning to do a release tonight if it's fine with you. Do you want to keep `Dictionary text font-family/font-size/line-height` as-is, add a comment, or mark it as missing for now?

jackiew1, 2022-04-15:
> @jackiew1, I'm planning to do a release in a few hours if it's fine with you. Do you want to keep `Dictionary text font-family/font-size/line-height` as-is, add a comment, or mark it as missing for now?

I think it may be a symptom of a bigger problem. The working KA1 I had now doesn't work either. In fact, I don't think any of the patches which have an option to customise font-family are working properly, i.e including nickel patches:
- `Custom header/footer captions`
- `Custom page navigation scrubber`
- `Customise Header back button`

The parts of these patches customising CSS properties other than font-family (e.g. font-size, line-height) are still working as far as I can tell.

I'm still looking at it, I'll get back to you but my gut feel is that if there is a solution it's not going to be found by me. Accessing fonts has obviously changed a lot maybe new techniques are required.

I think the only thing we can do is release the patches as-is but add a note to your new MR patch thread that customising font-family is no longer a working option in any of the 4 patches I've already mentioned today. I'll be asleep by the time you release them so I won't be able to write that note myself until tomorrow. If you want to do it yourself that's fine.

pgaskin, 2022-04-15:
> I think the only thing we can do is release the patches as-is but add a note to your new MR patch thread that customising font-family is no longer a working option in any of the 4 patches I've [...]

I'll add a comment to the patches, then release them in the next ~30 minutes. I'll wait a few minutes in case you want to write the comment yourself.

pgaskin, 2022-04-16:
Released: [`v75`](https://github.com/pgaskin/kobopatch-patches/releases/v75), MR:thread-346263.

pgaskin, 2022-04-16:
<details><summary><b>$ git diff --no-index src/versions/{4.31.19086,4.32.19501}</b></summary>

```diff
diff --git a/src/versions/4.31.19086/libadobe.so.yaml/pgaskin.yaml b/src/versions/4.32.19501/libadobe.so.yaml/pgaskin.yaml
index 81032e6..23d4c9b 100644
--- a/src/versions/4.31.19086/libadobe.so.yaml/pgaskin.yaml
+++ b/src/versions/4.32.19501/libadobe.so.yaml/pgaskin.yaml
@@ -6,5 +6,5 @@ Remove PDF map widget shown during panning:
   - BaseAddress:  {Sym: "N3AdobeReader::showMapWidget()"}
   # tail: N3AdobeReader::updatePanningMap() -> N3AdobeReader::hideMapWidget()
   # TODO: figure out what broke the plt parsing in kobopatch for libadobe in 18730+
-  - ReplaceBytes: {Offset:  80, FindInstBW: 0x1BDE8, ReplaceInstBW: 0x1B11C}
-  - ReplaceBytes: {Offset: 192, FindInstBW: 0x1BDE8, ReplaceInstBW: 0x1B11C}
+  - ReplaceBytes: {Offset:  80, FindInstBW: 0x1BF88, ReplaceInstBW: 0x1B2A4}
+  - ReplaceBytes: {Offset: 192, FindInstBW: 0x1BF88, ReplaceInstBW: 0x1B2A4}
diff --git a/src/versions/4.31.19086/libnickel.so.1.0.0.yaml/geoffr.yaml b/src/versions/4.32.19501/libnickel.so.1.0.0.yaml/geoffr.yaml
index 6c7b71c..1e70880 100644
--- a/src/versions/4.31.19086/libnickel.so.1.0.0.yaml/geoffr.yaml
+++ b/src/versions/4.32.19501/libnickel.so.1.0.0.yaml/geoffr.yaml
@@ -47,18 +47,24 @@ My 24 line spacing values:
     # basically rewrite the entire code starting where values are added:
     # save r5, initialise loop counter
   - BaseAddress:  {Sym: "ReadingSettings::lineHeightScalars() const", Rel: 84} # starting at first QList<double>::append(double const&)
-  - ReplaceBytes: {Offset:  0, FindInstBLX: {SymPLT: "QList<double>::append(double const&)"}, ReplaceH: A9 46 00 25}              # mov sb, r5       mov r5, #0
-  - ReplaceBytes: {Offset:  4, FindH: 6D A3, ReplaceH: ED 00}                         # lsls r5, r5, #3
+  - ReplaceBytes: {Offset:  0, FindInstBLX: {SymPLT: "QList<double>::append(double const&)"}, ReplaceH: 00 00 00 00} # clear out the BLX
+  - ReplaceBytes: {Offset: 20, FindInstBLX: {SymPLT: "QList<double>::append(double const&)"}, ReplaceH: 00 00 00 00} # clear out the BLX
+  - ReplaceBytes: {Offset:  0, FindH: 00 00, ReplaceH: A9 46} # mov  sb, r5
+  - ReplaceBytes: {Offset:  2, FindH: 00 00, ReplaceH: 00 25} # mov  r5, #0
+  - ReplaceBytes: {Offset:  4, FindH: 6C A3, ReplaceH: ED 00} # lsls r5, r5, #3
     # loop to load 14 vals from new table
-  - ReplaceBytes: {Offset:  6, FindH: D3 E9 00 23 07 F1, ReplaceH: 20 46 04 A1 29 44} # mov r0, r4   adr r1, #0x10  add r1, r5
-  - ReplaceBytes: {Offset: 12, FindH: 80 01 20 46,       ReplaceInstBLX: {SymPLT: "QList<double>::append(double const&)"}}
-  - ReplaceBytes: {Offset: 16, FindH: 61 E9 1C 23,       ReplaceH: 08 35 70 2D}       # add r5, #8   cmp r5, 112 ## NOTE: 14*8=112
-  - ReplaceBytes: {Offset: 20, FindInstBLX: {SymPLT: "QList<double>::append(double const&)"}, ReplaceH: 00 00 00 00}       # clear out the BLX in the way of the next two replacements
-  - ReplaceBytes: {Offset: 20, FindH: 00 00,             ReplaceH: F7 D1}             # bne #-10
-    # jump over new table
-  - ReplaceBytes: {Offset: 22, FindH: 00 00 6A A3 D3 E9, ReplaceH: 7C E0 00 BF 00 BF} # b #0xFC
+  - ReplaceBytes: {Offset:  6, FindH: D3 E9, ReplaceH: 20 46} # mov  r0, r4
+  - ReplaceBytes: {Offset:  8, FindH: 00 23, ReplaceH: 04 A1} # adr  r1, #0x10
+  - ReplaceBytes: {Offset: 10, FindH: 07 F1, ReplaceH: 29 44} # add  r1, r5
+  - ReplaceBytes: {Offset: 12, FindH: 80 01 20 46, ReplaceInstBLX: {SymPLT: "QList<double>::append(double const&)"}}
+  - ReplaceBytes: {Offset: 16, FindH: 61 E9, ReplaceH: 08 35} # add  r5, #8 (sizeof(double))
+  - ReplaceBytes: {Offset: 18, FindH: 1C 23, ReplaceH: 70 2D} # cmp  r5, 112 (14*sizeof(double))
+  - ReplaceBytes: {Offset: 20, FindH: 00 00, ReplaceH: F7 D1} # bne  #-10
+    # jump to the rest of the new code
+  - ReplaceBytes: {Offset: 22, FindH: 00 00, ReplaceH: 7B E0} # b #250 (356 − (84 + 22))
+  - ReplaceBytes: {Offset: 24, FindH: 69 A3 D3 E9, ReplaceInstNOP: true} # unused
     # new table: 14 8-aligned doubles
-  - BaseAddress:  {Sym: "ReadingSettings::lineHeightScalars() const", Rel: 112} # right after previous instruction
+  - BaseAddress:  {Sym: "ReadingSettings::lineHeightScalars() const", Rel: 112} # right after previous instruction (84 + 24 + 4)
   - ReplaceBytes: {Offset:  12, FindInstBLX: {SymPLT: "QList<double>::append(double const&)"}, ReplaceH: 00 00 00 00} # first, clear out the BLXs in the way
   - ReplaceBytes: {Offset:  32, FindInstBLX: {SymPLT: "QList<double>::append(double const&)"}, ReplaceH: 00 00 00 00}
   - ReplaceBytes: {Offset:  58, FindInstBLX: {SymPLT: "QList<double>::append(double const&)"}, ReplaceH: 00 00 00 00}
@@ -66,29 +72,33 @@ My 24 line spacing values:
   - ReplaceBytes: {Offset: 100, FindInstBLX: {SymPLT: "QList<double>::append(double const&)"}, ReplaceH: 00 00 00 00}
   - ReplaceBytes: {Offset:   0, FindH: 00 23 07 F1 80 01 20 46, ReplaceH: 00 00 00 00 00 00 F0 3F} # table of 1.0f values, to be replaced later
   - ReplaceBytes: {Offset:   8, FindH: 61 E9 1A 23 00 00 00 00, ReplaceH: 00 00 00 00 00 00 F0 3F}
-  - ReplaceBytes: {Offset:  16, FindH: 67 A3 D3 E9 00 23 07 F1, ReplaceH: 00 00 00 00 00 00 F0 3F}
+  - ReplaceBytes: {Offset:  16, FindH: 66 A3 D3 E9 00 23 07 F1, ReplaceH: 00 00 00 00 00 00 F0 3F}
   - ReplaceBytes: {Offset:  24, FindH: 80 01 20 46 61 E9 18 23, ReplaceH: 00 00 00 00 00 00 F0 3F}
   - ReplaceBytes: {Offset:  32, FindH: 00 00 00 00 43 F2 33 33, ReplaceH: 00 00 00 00 00 00 F0 3F}
[... cut ...]
       Replace: "#lblForgotPin,#lblSignOut{qproperty-visible:false;}"
@@ -23,7 +23,7 @@ Remove forgot pin button from lock screen:
 Increase size of kepub chapter progress chart:
   - Enabled: no
   - Description: Originally by oren64, rewritten for 4.16.13337 by pgaskin (geek1011).
-  - BaseAddress: 0x1667475 # qss/ReadingMenuStatsView.qss
+  - BaseAddress: 0x176E839 # qss/ReadingMenuStatsView.qss
   - ReplaceZlibGroup:
       Replacements:
       # Top padding is already set to 15px, 25px, 33px, vertical aligned to middle.
```

</details>

closed 2022-04-16 by pgaskin