pgaskin/kobopatch-patches #110: 3 Patch updates needed in librmsdk.so.1.0.0.yaml

jackiew1, 2022-08-22:
It appears these 3 patches no longer work:
- `Default ePub serif font (Amasis)`
- `Default ePub sans-serif font (Gill Sans)`
- `Default ePub monospace font`

Although they don't create errors when running `kobopatch` they are either no longer working at all (sans-serif) or no longer working as flexibly as they once did (serif, mono).

I suspect they stopped working in fw 4.32.19501 but no-one noticed until yesterday. So, obviously not used by many (any???).

I think they're all salvageable to a limited extent. I'm still doing some testing so I'll update them when I've finished. 

The serif one does still work as-is but the help notes need updating to explain the new limitations.

TL;DNR
Once upon a time you could be quite vague when specifying the new font-family name. As long as you got the first few chars right the Kobo could find a near match from its font list. Now you have to specify the exact whole font name. Trouble is there aren't many characters available to play with in the patches - serif/mono <=7-chars, sans <=11-chars (not too bad). Otherwise a font editor's needed to shorten the internal font name - enough to put off most users.

jackiew1, 2022-08-23:
@pgaskin I'm done. 

I suspect patch `Default ePub symbol font (Symbol)` would be similarly affected but as GeoffR didn't even know what it's supposed to do I've got no idea how to test it. A good candidate to be "quietly disappeared" if you ask me. 😃

pgaskin, 2022-08-23:
Thanks! I will probably backport your changes before the next release.

I suspect that patch would be for MathML characters, but I'm not even sure if RMSDK supports those. It might also be used as a fallback font too.

I'll probably just remove it as you recommend since it's extremely old and I don't think anybody uses it.

closed 2022-08-23 by pgaskin