pgaskin/kobopatch-patches #112: Patches for 4.34.20097

pgaskin, 2022-09-20:
- [ ] Upload testdata (@pgaskin)
- [ ] Extract resources (@pgaskin)
- [ ] Find libnickel offsets where necessary (@pgaskin)
- [ ] Find Clara 2E stuff (@pgaskin, @NiLuJe)
- [ ] Update most patches (@pgaskin)
- [ ] Update other patches if necessary (@pgaskin, @shermp)
- [ ] Update nickel patches if necessary (@jackiew1?)
- [ ] Test patches (@pgaskin)
- [ ] Make release (@pgaskin)
- [ ] MR post (@pgaskin)

pgaskin, 2022-09-20:
**Resource extraction:**

```bash
./qrc2zip --output "nickel.20097.qInitResources_styles.zip"       --recursive --verbose "nickel" 1 25236640 25101856 25225432 || exit 1
./qrc2zip --output "nickel.20097.qInitResources_resources.zip"    --recursive --verbose "nickel" 1 20455792    94856 20454448 || exit 1
./qrc2zip --output "nickel.20097.qInitResources_certificates.zip" --recursive --verbose "nickel" 1 25397936 25239776 25393760 || exit 1
./qrc2zip --output "nickel.20097.qInitResources_translations.zip" --recursive --verbose "nickel" 1 25101528 20456832 25100784 || exit 1
```

<details><summary><b>Output</b></summary>

```
nickel@20097/qInitResources_styles
DIR     "qss" (0x18114BC + 3108)
FILE    "qss/ManageBookWidget.qss" (0x17FC478 + 867)
FILE    "qss/ChooseSetupTypeView.qss" (0x17FA1BF + 615)
FILE    "qss/OverDriveSetupCompleteView.qss" (0x1806EBD + 473)
FILE    "qss/IconPowerView.qss" (0x1804CE7 + 661)
FILE    "qss/DemoRegionSelectionView.qss" (0x17F8FA0 + 608)
FILE    "qss/BTBSimilarLibraryView.qss" (0x17F6A81 + 618)
FILE    "qss/BookProgressBarWidget.qss" (0x1804541 + 408)
FILE    "qss/CheckBoxCaptionWidget.qss" (0x17F2B1A + 224)
FILE    "qss/SearchLanguageView.qss" (0x180D51D + 414)
FILE    "qss/QuickTurnSlider.qss" (0x18025E4 + 291)
FILE    "qss/N3BionicSettingsView.qss" (0x180C726 + 404)
FILE    "qss/AdvancedFontSettingsView.qss" (0x17F0A0E + 1107)
FILE    "qss/TouchTextEdit.qss" (0x1800350 + 992)
FILE    "qss/LibraryEmptyStateWidget.qss" (0x1804F80 + 344)
FILE    "qss/DictionaryActivityView.qss" (0x17F7855 + 516)
FILE    "qss/NotebookJitView.qss" (0x180E08D + 393)
FILE    "qss/AudiobookPlayer.qss" (0x17FCD5D + 1117)
FILE    "qss/NickelTouchMenu.qss" (0x18082F6 + 1006)
FILE    "qss/LibraryFinderView.qss" (0x180D485 + 110)
FILE    "qss/OverDriveHoldsStoreWidget.qss" (0x17F7A59 + 393)
FILE    "qss/BatteryMenuView.qss" (0x18050DC + 675)
FILE    "qss/N3SettingsLibraryView.qss" (0x1805EE9 + 568)
FILE    "qss/ReturningReaderDialog.qss" (0x180D7EC + 332)
FILE    "qss/WishlistView.qss" (0x180CD9F + 641)
FILE    "qss/AnnotationDetails.qss" (0x180628E + 975)
FILE    "qss/DetailedReviewViewHtml.qss" (0x17F848D + 827)
FILE    "qss/DictionarySearchItem.qss" (0x17F17A1 + 211)
FILE    "qss/PdfAnnotationItem.qss" (0x18013D1 + 319)
FILE    "qss/SkipWifiSetupDialog.qss" (0x17FAA04 + 622)
FILE    "qss/PocketFinishedMenuView.qss" (0x18080C7 + 555)
FILE    "qss/DragonMenuBarLabel.qss" (0x17F4126 + 668)
FILE    "qss/ParentalControlMenuView.qss" (0x180DCF2 + 664)
FILE    "qss/HtmlArticle.qss" (0x1801AFA + 2324)
FILE    "qss/StorefrontView.qss" (0x17F1F80 + 457)
FILE    "qss/ManageDictionariesView.qss" (0x17FDD33 + 364)
FILE    "qss/StoreNavTitleBar.qss" (0x1808C24 + 407)
FILE    "qss/DragonAnnotationView.qss" (0x17F826E + 543)
FILE    "qss/ReviewWidget.qss" (0x17FEB1C + 633)
FILE    "qss/SubscriptionUpgradeBanner.qss" (0x1808A82 + 414)
FILE    "qss/CategoriesListWidget.qss" (0x17FAC76 + 440)
FILE    "qss/ScrollBarPagingPopupStyle.qss" (0x18069A9 + 712)
FILE    "qss/HelpDialogView.qss" (0x1806121 + 27)
FILE    "qss/GenericHomeWidget.qss" (0x17F6CF3 + 537)
FILE    "qss/RotatePopup.qss" (0x180DF8E + 251)
FILE    "qss/MultiSelectTouchDropDown.qss" (0x17F3550 + 345)
FILE    "qss/ReadingMenuVolumeSettingsView.qss" (0x17F43C6 + 472)
FILE    "qss/AutoCompleteClear.qss" (0x1807B2C + 212)
FILE    "qss/ManageBooksView.qss" (0x17FD1BE + 477)
FILE    "qss/MyWordsView.qss" (0x180BBBB + 488)
FILE    "qss/DropboxView.qss" (0x17FF266 + 1172)
FILE    "qss/ExportFormatOptions.qss" (0x17FED99 + 647)
FILE    "qss/MainNavButton.qss" (0x17F2BFE + 302)
FILE    "qss/DragonLibraryView.qss" (0x17F0E69 + 892)
FILE    "qss/FramedDragonPowerView.qss" (0x18000DF + 625)
FILE    "qss/SelectionMenuView.qss" (0x1800738 + 473)
FILE    "qss/N3SettingsFrontlightLEDView.qss" (0x17F2929 + 489)
[... 2573 lines cut ...]
FILE    "default.rcc/images/mynotes/stroke_width_4_inverted.svg" (0x5C72C + 309)
FILE    "default.rcc/images/mynotes/stroke_width_5_inverted.svg" (0x5B265 + 309)
FILE    "default.rcc/images/mynotes/stroke_width_3_inverted.svg" (0x5CB5C + 309)
FILE    "default.rcc/images/mynotes/stroke_width_2_inverted.svg" (0x5DE2A + 309)
FILE    "default.rcc/images/mynotes/stroke_width_1_inverted.svg" (0x5F058 + 292)
FILE    "default.rcc/images/mynotes/stroke_width_2.svg" (0x5F486 + 265)
FILE    "default.rcc/images/mynotes/stroke_width_4.svg" (0x5EA4B + 265)
FILE    "default.rcc/images/mynotes/stroke_width_3.svg" (0x5F180 + 265)
FILE    "default.rcc/images/mynotes/stroke_width_5.svg" (0x5D006 + 265)
FILE    "default.rcc/images/mynotes/stroke_width_1.svg" (0x5FFE3 + 248)
FILE    "default.rcc/images/mynotes/arrow_right_thin.svg" (0x5EB58 + 320)
FILE    "default.rcc/images/mynotes/math.svg" (0x5B9EA + 414)
FILE    "default.rcc/images/mynotes/diagram.svg" (0x5D5BA + 882)
FILE    "default.rcc/images/mynotes/underline.svg" (0x5DF63 + 235)
FILE    "default.rcc/images/mynotes/calligraphy_pen_inverted.svg" (0x5CC95 + 877)
FILE    "default.rcc/images/mynotes/ballpoint_pen.svg" (0x5B48F + 1105)
FILE    "default.rcc/images/mynotes/eraser.svg" (0x5C983 + 469)
FILE    "default.rcc/images/mynotes/stylus_swipe_down.svg" (0x5B8E4 + 258)
FILE    "default.rcc/images/mynotes/tap_hand.svg" (0x5A670 + 833)
FILE    "default.rcc/images/mynotes/freeform.svg" (0x5BE96 + 522)
FILE    "default.rcc/images/mynotes/brush_inverted.svg" (0x5EC9C + 952)
FILE    "default.rcc/images/mynotes/redo.svg" (0x5D113 + 429)
FILE    "default.rcc/images/mynotes/brush.svg" (0x5F593 + 941)
FILE    "default.rcc/images/mynotes/fountain_pen_inverted.svg" (0x5AD3F + 1314)
FILE    "default.rcc/image

pgaskin, 2022-09-20:
**Clara 2E**

- goldfinch
- TODO

**Misc**

- TODO

NiLuJe, 2022-09-20:
Startup scripts & udev changes appear minimal:

```diff
diff -x ntx_hwconfig -Nuarp 4.33/ROOT/bin/kobo_config.sh 4.34/ROOT/bin/kobo_config.sh
--- 4.33/ROOT/bin/kobo_config.sh        2022-07-21 00:13:58.000000000 +0200
+++ 4.34/ROOT/bin/kobo_config.sh        2022-09-16 15:12:44.000000000 +0200
@@ -27,5 +27,6 @@ case $PCB in
      EA0P1*) echo europa;;
      E70K1*) echo io;;
      E80P0*) echo cadmus;;
+     E60K2*) echo goldfinch;;
      *) echo trilogy;;
 esac
diff -x ntx_hwconfig -Nuarp 4.33/ROOT/etc/init.d/rcS 4.34/ROOT/etc/init.d/rcS
--- 4.33/ROOT/etc/init.d/rcS 2022-07-21 00:13:58.000000000 +0200
+++ 4.34/ROOT/etc/init.d/rcS 2022-09-16 15:12:44.000000000 +0200
@@ -26,8 +26,7 @@ onboard_partition() {
 write_uboot_env() {
      if [ $1 == freescale ] || [ $1 == mx50-ntx ] || [ $1 == mx6sl-ntx ] || [ $1 == mx6sll-ntx ] || [ $1 == mx6ull-ntx ]; then
       dd if=$2 of=/dev/mmcblk0 bs=128k count=1 seek=6
-     fi
-     if [ $1 == b300-ntx ]; then
+     elif [ $1 == b300-ntx ]; then
       dd if=$2 of=/dev/mmcblk0 bs=512 seek=1286
      fi
      sync
@@ -97,8 +96,11 @@ else
      tar zxf /etc/udev.tgz -C /
 fi
 
-export UBOOT_MMC=/etc/u-boot/$PLATFORM/u-boot.mmc
-export UBOOT_RECOVERY=/etc/u-boot/$PLATFORM/u-boot.recovery
+
+UBOOT_MMC=/etc/u-boot/$PLATFORM/u-boot.mmc
+UBOOT_RECOVERY=/etc/u-boot/$PLATFORM/u-boot.recovery
+export UBOOT_MMC
+export UBOOT_RECOVERY
 
 find /usr/local/Trolltech/QtEmbedded-4.6.2-arm/lib/ -type f -exec readahead {} \;
 
@@ -123,6 +125,8 @@ if [ $? != 0 ] || [ $FS_CORRUPT == 1 ] |
        export COORDINATES="320 1400 730 170";;
       nova)
        export COORDINATES="250 1070 600 150";;
+      goldfinch)
+       export COORDINATES="250 1070 600 150";;
       frost)
        export COORDINATES="350 1400 750 200";;
       storm)
diff -x ntx_hwconfig -Nuarp 4.33/ROOT/etc/init.d/upgrade-wifi.sh 4.34/ROOT/etc/init.d/upgrade-wifi.sh
--- 4.33/ROOT/etc/init.d/upgrade-wifi.sh        2020-09-10 15:45:09.000000000 +0200
+++ 4.34/ROOT/etc/init.d/upgrade-wifi.sh        2022-09-16 15:12:44.000000000 +0200
@@ -1,5 +1,17 @@
 #!/bin/sh
 
+platform_detect() {
+     PLATFORM=freescale
+     if [ `dd if=/dev/mmcblk0 bs=512 skip=1024 count=1 | grep -c "HW CONFIG"` == 1 ]; then
+      CPU=`ntx_hwconfig -s -p /dev/mmcblk0 CPU`
+      PLATFORM=$CPU-ntx
+      WIFI=`ntx_hwconfig -s -p /dev/mmcblk0 Wifi`
+      PCB=`ntx_hwconfig -s -p /dev/mmcblk0 PCB`
+      PCB_LVL=`ntx_hwconfig -s -p /dev/mmcblk0 PCB_LVL`
+      PCB_REV=`ntx_hwconfig -s -p /dev/mmcblk0 PCB_REV`
+     fi
+}
+
 if [ -e /mnt/onboard/.kobo/manifest.md5sum ]; then 
      cd /mnt/onboard/.kobo
      md5sum -c manifest.md5sum
[... 85 lines cut ...]
+
+# by-path (parent device path)
+ENV{DEVTYPE}=="disk", IMPORT{program}="path_id %p"
+ENV{DEVTYPE}=="disk", ENV{ID_PATH}=="?*", SYMLINK+="disk/by-path/$env{ID_PATH}"
+ENV{DEVTYPE}=="partition", ENV{ID_PATH}=="?*", SYMLINK+="disk/by-path/$env{ID_PATH}-part%n"
+
+# skip unpartitioned removable media devices from drivers which do not send "change" events
+ENV{DEVTYPE}=="disk", KERNEL!="sd*|sr*", ATTR{removable}=="1", GOTO="persistent_storage_end"
+
+# probe filesystem metadata of optical drives which have a media inserted
+KERNEL=="sr*", ENV{ID_CDROM_MEDIA}=="?*", IMPORT{program}="/sbin/blkid -o udev -p -u noraid -O $env{ID_CDROM_MEDIA_SESSION_LAST_OFFSET} $tempnode"
+
+# probe filesystem metadata of disks
+KERNEL!="sr*", IMPORT{program}="/sbin/blkid -o udev -p $tempnode"
+
+# watch for future changes
+KERNEL!="sr*", OPTIONS+="watch"
+
+# by-label/by-uuid links (filesystem metadata)
+ENV{ID_FS_USAGE}=="filesystem|other|crypto", ENV{ID_FS_UUID_ENC}=="?*", SYMLINK+="disk/by-uuid/$env{ID_FS_UUID_ENC}"
+ENV{ID_FS_USAGE}=="filesystem|other", ENV{ID_FS_LABEL_ENC}=="?*", SYMLINK+="disk/by-label/$env{ID_FS_LABEL_ENC}"
+ENV{ID_PART_ENTRY_SCHEME}=="gpt", ENV{ID_PART_ENTRY_NAME}=="?*", SYMLINK+="disk/by-partlabel/$env{ID_PART_ENTRY_NAME}"
+
+LABEL="persistent_storage_end"
```

NiLuJe, 2022-09-20:
That's for a kobo8 vs. kobo8, FWIW.

I think I've seen that `platform_detect` change before on the Libra 2 release, IIRC.

Don't recall anything about the `60-persistent-storage.rules`, but it's not there on my Elipsa, at least ;).

pgaskin, 2022-09-20:
```
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.34.20097/nickel.yaml/pgaskin.yaml: Increase size of kepub chapter progress chart: could not apply patch "Increase size of kepub chapter progress chart": line 28: inst 4: ReplaceZlib: not a zlib stream
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.34.20097/nickel.yaml/pgaskin.yaml: Remove forgot pin button from lock screen: could not apply patch "Remove forgot pin button from lock screen": line 20: inst 4: ReplaceZlib: not a zlib stream
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.34.20097/libnickel.so.1.0.0.yaml/pgaskin.yaml: Customize ComfortLight settings: could not apply patch "Customize ComfortLight settings": line 535: inst 4: ReplaceInt: could not find specified bytes at offset
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.34.20097/libnickel.so.1.0.0.yaml/pgaskin.yaml: Hide browser from beta features: could not apply patch "Hide browser from beta features": line 82: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.34.20097/libnickel.so.1.0.0.yaml/pgaskin.yaml: Never show Kobo Plus, wishlist, and points SmartLinks: could not apply patch "Never show Kobo Plus, wishlist, and points SmartLinks": line 288: inst 5: ReplaceBytes: could not find specified bytes at offset
```

Seems like it's just the usual, thankfully.

pgaskin, 2022-09-21:
Nickel patches should be fine.

jackiew1, 2022-09-21:
> Nickel patches should be fine.

All the ones I use seem OK as-is.

`qApp_deviceIsGoldfinch=true` is only used twice in nickel qss files. Both Bluetooth-related so not involved in any current patches.

There's a couple of new qss streams for the new 3-book `QuickAccess` feature but, off-hand, I didn't see anything which would be interesting for a new patch.

There's also a new qss called `N3BionicSettingsView.qss`. I didn't see anything in the release notes which sounded related.

I guess the Clara2e needs to be added to all the patch notes in the yaml files. Without one to play with I'm not sure where it belongs. If I had to guess I'd say it should be added to each patch wherever the current ClaraHD is located, because it's the same screen-size/resolution (I think?) If you want me to do them all it it will have to be tomorrow as it's nearly 2am and I have to be up early. Bye for now.

pgaskin, 2022-09-21:
I'm planning to release the patches tonight if nothing goes wrong during testing, possibly with a new patch for disabling quick access. I'll leave updating the device lists for later when I have more time to go through and make sure it's correct and any exceptions are accounted for.

pgaskin, 2022-09-21:
<details><summary>Symbols</summary>

```
							      >	BROWSE_HISTORY_TAB_ID
							      >	_ZN10SyncClient21addStoreFrontCommandsEv
_ZN10VirtualKey13setWidthScaleEf			      <
							      >	_ZN10VirtualKey16setLayoutStretchEi
							      >	_ZN11ReadingView15openQuickAccessEv
							      >	_ZN12WebRequester21onestoreBrowseHistoryERK4UserP19WebRespons
							      >	_ZN13EventInfoImpl17quickAccessOpenedEv
							      >	_ZN13PopupKeyboard10layoutKeysEv
							      >	_ZN13PowerSettings22setRecentlyViewEnabledEb
							      >	_ZN13PowerSettings24getRecentlyViewedEnabledEv
							      >	_ZN13StoreListView15updateTabWidgetERK3Tab
							      >	_ZN13StoreListView22updatePromoPlaceholderERK3Tab
							      >	_ZN14OppositeFilterI20VolumeIDSearchFilter6VolumeE6filterERKS
							      >	_ZN14OppositeFilterI20VolumeIDSearchFilter6VolumeED0Ev
							      >	_ZN14OppositeFilterI20VolumeIDSearchFilter6VolumeED1Ev
							      >	_ZN14OppositeFilterI20VolumeIDSearchFilter6VolumeED2Ev
							      >	_ZN14QSharedPointerI20ScopedMainWindowHideE5derefEPN15QtShare
							      >	_ZN15QNetworkRequest12setAttributeENS_9AttributeERK8QVariant
							      >	_ZN15QtSharedPointer33ExternalRefCountWithCustomDeleterI20Sco
							      >	_ZN15ReadingMenuView15openQuickAccessEv
							      >	_ZN15SmartLinkWidget23onSubscriptionPurchasedEv
							      >	_ZN15StoreListWidget19setPromoPlaceholderEv
							      >	_ZN15SyncManagerBase20createDemoSyncWorkerERK6Device
							      >	_ZN15VirtualKeyboard21setKeyboardMaxColumnsEi
							      >	_ZN16DragonListWidget11disableMenuEv
							      >	_ZN17SubscriptionMixin13learnMoreFromE7QString
_ZN17SubscriptionMixin9learnMoreEv			      <
							      >	_ZN18QuickAccessMiniBar10tapGestureEP15GestureReceiverP10TapG
							      >	_ZN18QuickAccessMiniBar11qt_metacallEN11QMetaObject4CallEiPPv
							      >	_ZN18QuickAccessMiniBar11qt_metacastEPKc
							      >	_ZN18QuickAccessMiniBar12swipeGestureEP15GestureReceiverP12Sw
							      >	_ZN18QuickAccessMiniBar13defaultColorsEv
							      >	_ZN18QuickAccessMiniBar13handleGestureEP8QGesture
							      >	_ZN18QuickAccessMiniBar13reverseColorsEb
							      >	_ZN18QuickAccessMiniBar13reverseColorsEv
							      >	_ZN18QuickAccessMiniBar15openQuickAccessEv
							      >	_ZN18QuickAccessMiniBar16staticMetaObjectE
							      >	_ZN18QuickAccessMiniBar5eventEP6QEvent
							      >	_ZN18QuickAccessMiniBar6tappedEb
							      >	_ZN18QuickAccessMiniBar9mouseDownEv
							      >	_ZN18QuickAccessMiniBarC1EP7QWidget
							      >	_ZN18QuickAccessMiniBarC2EP7QWidget
							      >	_ZN18QuickAccessMiniBarD0Ev
							      >	_ZN18QuickAccessMiniBarD1Ev
							      >	_ZN18QuickAccessMiniBarD2Ev
							      >	_ZN19ApplicationSettings13is24HourClockEv
							      >	_ZN19ApplicationSettings14set24HourClockEb
							      >	_ZN19N3SettingsPowerView24setRecentlyViewedEnabledEb
							      >	_ZN19PreviewActionWidget23onSubscriptionPurchasedEv
							      >	_ZN19QuickAccessMenuView11qt_metacallEN11QMetaObject4CallEiPP
							      >	_ZN19QuickAccessMenuView11qt_metacastEPKc
							      >	_ZN19QuickAccessMenuView12swipeGestureEP15GestureReceiverP12S
							      >	_ZN19QuickAccessMenuView16staticMetaObjectE
							      >	_ZN19QuickAccessMenuView17setupViewGeometryEv
							      >	_ZN19QuickAccessMenuView5eventEP6QEvent
							      >	_ZN19QuickAccessMenuView9closeMenuEv
							      >	_ZN19QuickAccessMenuViewC1EP7QWidgetRK6Volume
							      >	_ZN19QuickAccessMenuViewC2EP7QWidgetRK6Volume
							      >	_ZN19QuickAccessMenuViewD0Ev
							      >	_ZN19QuickAccessMenuViewD1Ev
							      >	_ZN19QuickAccessMenuViewD2Ev
							      >	_ZN19StoreListController9updateTabERK3Tab
							      >	_ZN20BlueZServiceProviderD0Ev
							      >	_ZN20BlueZServiceProviderD1Ev
							      >	_ZN20BlueZServiceProviderD2Ev
							      >	_ZN20KoboPlusQueueCommand13assetsChangedEv
							      >	_ZN20KoboPlusQueueCommand6finishEv
[... 206 lines cut ...]
							      >	_ZTS25QuickAccessMenuController
							      >	_ZTS27BroadcastAgentManager1Proxy
							      >	_ZTS28GridAndListLibraryControllerI22QuickAccessLibraryViewE
							      >	_ZTS28OnestoreBrowseHistoryRequest
							      >	_ZTS28QuickAccessLibraryController
							      >	_ZTS30QuickAccessLibraryDataProvider
							      >	_ZTS33BroadcastAgentManager1ProxyFacade
							      >	_ZTV14OppositeFilterI20VolumeIDSearchFilter6VolumeE
							      >	_ZTV18QuickAccessMiniBar
							      >	_ZTV19QuickAccessMenuView
							      >	_ZTV20BlueZServiceProvider
							      >	_ZTV20VolumeIDSearchFilter
							      >	_ZTV22BroadcastAgent1Adaptor
							      >	_ZTV22QuickAccessLibraryView
							      >	_ZTV24BluetoothServiceProvider
							      >	_ZTV25QuickAccessMenuController
							      >	_ZTV27BroadcastAgentManager1Proxy
							      >	_ZTV28GridAndListLibraryControllerI22QuickAccessLibraryViewE
							      >	_ZTV28OnestoreBrowseHistoryRequest
							      >	_ZTV28QuickAccessLibraryController
							      >	_ZTV30QuickAccessLibraryDataProvider
							      >	_ZTV33BroadcastAgentManager1ProxyFacade
```

</details>

pgaskin, 2022-09-21:
Released: [`v79`](https://github.com/pgaskin/kobopatch-patches/releases/v79), MR:thread-349230.

pgaskin, 2022-09-21:
<details><summary><b>$ git diff v78:src/versions/4.33.19759 v79:src/versions/4.34.20097</b></summary>

```diff
diff --git a/libnickel.so.1.0.0.yaml/pgaskin.yaml b/libnickel.so.1.0.0.yaml/pgaskin.yaml
index 8ce8b26..b7a7817 100644
--- a/libnickel.so.1.0.0.yaml/pgaskin.yaml
+++ b/libnickel.so.1.0.0.yaml/pgaskin.yaml
@@ -80,7 +80,7 @@ Hide browser from beta features:
   - Description: Hides the built-in browser from beta features.
   - ReplaceBytes:
       Base:        "N3SettingsExtrasView::N3SettingsExtrasView(QWidget*)"
-      Offset:      1372
+      Offset:      1392
       FindInstBLX: {SymPLT: "Device::isParentalControlEnabled() const"}
       ReplaceH:    4F F0 01 00
   - ReplaceBytes:
@@ -232,22 +232,22 @@ Set visible SmartLink:
   #  7 - KoboPlusFTE     - kobo plus enabled && not subscribed                 - KOBO PLUS: Browse Kobo Plus books                                - kobo store -> kobo plus
   #  8 - OverDrive       - overdrive enabled && signed in && have hold         - OVERDRIVE: View your public library holds                        - overdrive holds list
   # ROTATION MESSAGES (randomly chosen): (SmartLinkWidget::showRotationMessage, ActivityManager::smartLinkGeneralMessage, SmartLinkWidget::tapped)
-  #  1 - Pocket          - pocket not signed in                                - POCKET: Read articles from the web on your eReader               - pocket about
-  #  2 - KoboPlus        - kobo plus enabled && not subscribed                 - KOBO PLUS: Get unlimited access to books in all genres           - kobo store -> kobo plus
-  #  2 - KoboPlus        - kobo plus enabled && not subscribed                 - KOBO PLUS: Read as much as you want with a Kobo Plus subscription- kobo store -> kobo plus
-  #  3 - Overdrive       - overdrive enabled && not signed in                  - OVERDRIVE: Borrow eBooks from your public library                - overdrive about
-  #  4 - Categories      - always                                              - CATEGORIES: Browse fiction, romance, biography and more          - kobo store -> categories
-  #  9 - NaturalLight    - has light sensor && not viewed tutorial             - DEVICE NAME: Learn about the Natural Light feature               - rgb front light about
-  # 10 - QuickTour       - quick tour not shown && quick turn tile not visible - GETTING STARTED: Get to know your %0                             - quick tour
-  # 11 - ReadingTour     - read a book tour not showed                         - READING A BOOK: Get quick tips about reading on your eReader     - read a book tutorial
-  # 12 - UserGuide       - user guide present && not opened                    - USER GUIDE: Read the user guide for your %0                      - read -> user guide
-  # 13 - RelatedReads    - recent book available from Activity                 - RELATED READS: Discover books related to the ones you're reading - browse -> similar books
-  # 14 - Wishlist        - no wishlist items || something else                 - WISHLIST: Create a Wishlist of books you're interested in        - browse -> wishlist
-  # 15 - ReadingSettings - reading settings not changed                        - READING SETTINGS: Customize the way you read a book              - settings -> reading settings
-  # 16 - ReadingStats    - books finished > 0 && hours read rounded to .5 > 0  - READING STATS: You've finished %n books and read for %n hours    - reading life -> stats
-  # 17 - SuperPoints     - participating in super points && points > 2400      - KOBO SUPER POINTS: You have %0 Super Points to redeem            - kobo store
-  # 18 - PocketUser      - unread pocket articles > 0                          - POCKET: Catch up on the %n article(s) you added recently         - library -> pocket
-  # 19 - Audiobooks      - supports audiobooks && audiobooks enabled           - AUDIOBOOKS: Get started with Audiobooks on your eReader          - kobo store -> audiobooks
+  #  1 - Pocket          - pocket not signed in                                - POCKET: Read articles from the web on your eReader                - pocket about
+  #  2 - KoboPlus        - kobo plus enabled && not subscribed                 - KOBO PLUS: Upgrade to a Kobo Plus Read & Listen subscription      - kobo store -> kobo plus
+  #  2 - KoboPlus        - kobo plus enabled && not subscribed                 - KOBO PLUS: Read as much as you want with a Kobo Plus subscription - kobo store -> kobo plus
+  #  3 - Overdrive       - overdrive enabled && not signed in                  - OVERDRIVE: Borrow eBooks from your public library                 - overdrive about
+  #  4 - Categories      - always                                              - CATEGORIES: Browse fiction, romance, biography and more           - kobo store -> categories
+  #  9 - NaturalLight    - has light sensor && not viewed tutorial             - DEVICE NAME: Learn about the Natural Light feature                - rgb front light about
+  # 10 - QuickTour       - quick tour not shown && quick turn tile not visible - GETTING STARTED: Get to know your %0                              - quick tour
+  # 11 - ReadingTour     - read a book tour not showed                         - READING A BOOK: Get quick tips about reading on your eReader      - read a book tutorial
+  # 12 - UserGuide       - user guide present && not opened                    - USER GUIDE: Read the user guide for your %0                       - read -> user guide
[... cut ...]
       Replace: "#lblForgotPin,#lblSignOut{qproperty-visible:false;}"
@@ -23,7 +23,7 @@ Remove forgot pin button from lock screen:
 Increase size of kepub chapter progress chart:
   - Enabled: no
   - Description: Originally by oren64, rewritten for 4.16.13337 by pgaskin (geek1011).
-  - BaseAddress: 0x17EF921 # qss/ReadingMenuStatsView.qss
+  - BaseAddress: 0x17F4EBE # qss/ReadingMenuStatsView.qss
   - ReplaceZlibGroup:
       Replacements:
       # Top padding is already set to 15px, 25px, 33px, vertical aligned to middle.
```

</details>

closed 2022-09-21 by pgaskin