pgaskin/kobopatch-patches #113: Update nickel patch `New home screen subtitle custom font`

jackiew1, 2022-10-12:
[See this MR thread.](https://www.mobileread.com/forums/showthread.php?t=349523)

Kobo added UI support for Polish in fw 4.34.20097 but it seems that the Home screen display was not tested well enough because the font-size used for the '% Read' caption on the top row is too big to accommodate the Polish equivalent, e.g. it shows as `Przeczytano 28%` rather than the English `28% Read`, so the important bit, `28%`, falls off the right edge.

This is a simple fix by including extra options to change font-size for the upper caption as well as the existing lower caption (time remaining).

jackiew1, 2022-10-12:
@pgaskin, I've committed the change. You can close whenever you want.

pgaskin, 2022-10-13:
Thanks! Note that for single-commit fixes like this, you can include `closes #113` or `fixes #113` somewhere in the commit message (I usually put it like ` (fixes #113)` at the end of the title, or `fixes #113` after a blank link at the end of the body.

I'll do a release with this change either a month from now or when the next update gets released.

closed 2022-10-13 by pgaskin

jackiew1, 2022-10-13:
> Thanks! Note that for single-commit fixes like this, you can include `closes #113` or `fixes #113` somewhere in the commit message (I usually put it like ` (fixes #113)` at the end of the title, or `fixes #113` after a blank link at the end of the body.

I've always thought you preferred to close issues yourself. That's why I didn't. Is that not the case?

> I'll do a release with this change either a month from now or when the next update gets released.

Unless you know something I don't, there's no need to do a new release before the next fw update. Only one person reported the problem and they've already implemented the updated patch. If I see someone else raising the same issue I'll direct them to the MR thread where I posted the new patch code.

Apparently the Polish problem has already been reported to Kobo so they may apply their own fix for next time. In which case I may need to rejig the patch again anyway.