pgaskin/kobopatch-patches #114: Patches for 4.35.20400

pgaskin, 2022-12-08:
- [ ] Upload testdata (@pgaskin)
- [ ] Extract resources (@pgaskin)
- [ ] Find libnickel offsets where necessary (@pgaskin)
- [ ] Update most patches (@pgaskin)
- [ ] Update other patches if necessary (@pgaskin, @shermp)
- [ ] Update nickel patches if necessary (@jackiew1?)
- [ ] Test patches (@pgaskin)
- [ ] Make release (@pgaskin)
- [ ] MR post (@pgaskin)
- Later (from #112): Add Clara 2E to koboutils (@pgaskin)

pgaskin, 2022-12-08:
**Resource extraction:**

```bash
python scripts/armqrc.py ~/kp/v/20400/nickel 
/home/patrick/kp/v/20400/nickel 1 25245712 25110856 25234504 # styles
/home/patrick/kp/v/20400/nickel 1 20458184    95696 20456880 # resources
/home/patrick/kp/v/20400/nickel 1 25407008 25248848 25402832 # certificates
/home/patrick/kp/v/20400/nickel 1 25110528 20459208 25109784 # translations
```

```bash
./qrc2zip --output "nickel.20400.qInitResources_styles.zip"       --recursive --verbose "nickel" 1 25245712 25110856 25234504 || exit 1
./qrc2zip --output "nickel.20400.qInitResources_resources.zip"    --recursive --verbose "nickel" 1 20458184    95696 20456880 || exit 1
./qrc2zip --output "nickel.20400.qInitResources_certificates.zip" --recursive --verbose "nickel" 1 25407008 25248848 25402832 || exit 1
./qrc2zip --output "nickel.20400.qInitResources_translations.zip" --recursive --verbose "nickel" 1 25110528 20459208 25109784 || exit 1
```

<details><summary><b>Output</b></summary>

```
nickel@20400/qInitResources_styles
DIR     "qss" (0x181382C + 3108)
FILE    "qss/ManageBookWidget.qss" (0x18038E3 + 867)
FILE    "qss/ChooseSetupTypeView.qss" (0x1806DF1 + 615)
FILE    "qss/OverDriveSetupCompleteView.qss" (0x17F7A0A + 473)
FILE    "qss/IconPowerView.qss" (0x17F6432 + 661)
FILE    "qss/DemoRegionSelectionView.qss" (0x1809903 + 608)
FILE    "qss/BTBSimilarLibraryView.qss" (0x17F6EFC + 618)
FILE    "qss/BookProgressBarWidget.qss" (0x180374B + 408)
FILE    "qss/CheckBoxCaptionWidget.qss" (0x17F400C + 224)
FILE    "qss/SearchLanguageView.qss" (0x1800D61 + 414)
FILE    "qss/QuickTurnSlider.qss" (0x1803624 + 291)
FILE    "qss/N3BionicSettingsView.qss" (0x17F72CA + 474)
FILE    "qss/AdvancedFontSettingsView.qss" (0x17F2C6A + 1107)
FILE    "qss/TouchTextEdit.qss" (0x180FAEC + 992)
FILE    "qss/LibraryEmptyStateWidget.qss" (0x17F716E + 344)
FILE    "qss/DictionaryActivityView.qss" (0x180A518 + 516)
FILE    "qss/NotebookJitView.qss" (0x17F418C + 393)
FILE    "qss/AudiobookPlayer.qss" (0x17FEE2D + 1117)
FILE    "qss/NickelTouchMenu.qss" (0x180D079 + 1006)
FILE    "qss/LibraryFinderView.qss" (0x18021F8 + 110)
FILE    "qss/OverDriveHoldsStoreWidget.qss" (0x17F37EA + 393)
FILE    "qss/BatteryMenuView.qss" (0x1808372 + 675)
FILE    "qss/N3SettingsLibraryView.qss" (0x1806133 + 568)
FILE    "qss/ReturningReaderDialog.qss" (0x18105E9 + 332)
FILE    "qss/WishlistView.qss" (0x18056A3 + 641)
FILE    "qss/AnnotationDetails.qss" (0x17FACC3 + 975)
FILE    "qss/DetailedReviewViewHtml.qss" (0x180C0FB + 827)
FILE    "qss/DictionarySearchItem.qss" (0x17FABEC + 211)
FILE    "qss/PdfAnnotationItem.qss" (0x17F4955 + 319)
FILE    "qss/SkipWifiSetupDialog.qss" (0x180749F + 622)
FILE    "qss/PocketFinishedMenuView.qss" (0x17FE93F + 555)
FILE    "qss/DragonMenuBarLabel.qss" (0x1805261 + 668)
FILE    "qss/ParentalControlMenuView.qss" (0x1800F03 + 664)
FILE    "qss/HtmlArticle.qss" (0x17FBB8C + 2324)
FILE    "qss/StorefrontView.qss" (0x180032D + 457)
FILE    "qss/ManageDictionariesView.qss" (0x1810739 + 364)
FILE    "qss/StoreNavTitleBar.qss" (0x17F542A + 407)
FILE    "qss/DragonAnnotationView.qss" (0x1800819 + 543)
FILE    "qss/ReviewWidget.qss" (0x17F4A98 + 633)
FILE    "qss/SubscriptionUpgradeBanner.qss" (0x1805501 + 414)
FILE    "qss/CategoriesListWidget.qss" (0x1801BDA + 440)
FILE    "qss/ScrollBarPagingPopupStyle.qss" (0x18027E3 + 712)
FILE    "qss/HelpDialogView.qss" (0x17F56CC + 27)
FILE    "qss/GenericHomeWidget.qss" (0x1809EE3 + 537)
FILE    "qss/RotatePopup.qss" (0x180B2E3 + 251)
FILE    "qss/MultiSelectTouchDropDown.qss" (0x17F47F4 + 345)
FILE    "qss/ReadingMenuVolumeSettingsView.qss" (0x17FF28E + 472)
FILE    "qss/AutoCompleteClear.qss" (0x17FDE3F + 212)
FILE    "qss/ManageBooksView.qss" (0x17FEC4C + 477)
[... 2578 lines cut ...]
FILE    "default.rcc/images/mynotes/drawing.svg" (0x59101 + 2549)
FILE    "default.rcc/images/mynotes/highlighter.svg" (0x57D99 + 774)
FILE    "default.rcc/images/mynotes/calligraphy_pen.svg" (0x56723 + 833)
FILE    "default.rcc/images/mynotes/pressure.svg" (0x5415A + 869)
FILE    "default.rcc/images/mynotes/menu_up_chevron.svg" (0x552D4 + 238)
FILE    "default.rcc/images/mynotes/menu_down_chevron_hit.svg" (0x5795F + 281)
FILE    "default.rcc/images/mynotes/undo.svg" (0x553C6 + 425)
FILE    "default.rcc/images/mynotes/stroke_width_4_inverted.svg" (0x587B6 + 309)
FILE    "default.rcc/images/mynotes/stroke_width_5_inverted.svg" (0x5519B + 309)
FILE    "default.rcc/images/mynotes/stroke_width_3_inverted.svg" (0x59AFA + 309)
FILE    "default.rcc/images/mynotes/stroke_width_2_inverted.svg" (0x54021 + 309)
FILE    "default.rcc/images/mynotes/stroke_width_1_inverted.svg" (0x54D2E + 292)
FILE    "default.rcc/images/mynotes/stroke_width_2.svg" (0x58254 + 265)
FILE    "default.rcc/images/mynotes/stroke_width_4.svg" (0x56B57 + 265)
FILE    "default.rcc/images/mynotes/stroke_width_3.svg" (0x573D1 + 265)
FILE    "default.rcc/images/mynotes/stroke_width_5.svg" (0x56408 + 265)
FILE    "default.rcc/images/mynotes/stroke_width_1.svg" (0x58AF6 + 248)
FILE    "default.rcc/images/mynotes/arrow_right_thin.svg" (0x57A7C + 320)
FILE    "default.rcc/images/mynotes/math.svg" (0x59C33 + 414)
FILE    "default.rcc/images/mynotes/diagram.svg" (0x56092 + 882)
FILE    "default.rcc/images/mynotes/underline.svg" (0x56A68 + 235)
FILE    "default.rcc/images/mynotes/calligraphy_pen_inverted.svg" (0x544C3 + 877)
FILE    "default.rcc/images/mynotes/ballpoint_pen.svg" (0x58361 + 1105)
FILE    "default.rcc/images/mynotes/eraser.svg" (0x58F28 + 469)
FILE    "default.rcc/images/mynotes/stylus_swipe_down.

pgaskin, 2022-12-08:
```
Errors:
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.35.20400/libadobe.so.yaml/pgaskin.yaml: Remove PDF map widget shown during panning: could not apply patch "Remove PDF map widget shown during panning": line 9: inst 4: ReplaceBytes: could not find specified bytes
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.35.20400/libnickel.so.1.0.0.yaml/pgaskin.yaml: Customize ComfortLight settings: could not apply patch "Customize ComfortLight settings": line 535: inst 4: ReplaceInt: could not find specified bytes at offset
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.35.20400/libnickel.so.1.0.0.yaml/pgaskin.yaml: Both page turn sides go next: could not apply patch "Both page turn sides go next": line 17: inst 4: ReplaceBytes: could not find specified bytes
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.35.20400/nickel.yaml/pgaskin.yaml: Remove forgot pin button from lock screen: could not apply patch "Remove forgot pin button from lock screen": line 20: inst 4: ReplaceZlib: not a zlib stream
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.35.20400/nickel.yaml/pgaskin.yaml: Increase size of kepub chapter progress chart: could not apply patch "Increase size of kepub chapter progress chart": line 28: inst 4: ReplaceZlib: not a zlib stream
```

Doesn't seem too terrible; libadobe will need a little bit of work, but libnickel and nickel are pretty much normal.

I don't think there will be much, if anything for @jackiew1 this release given that the only change is really just the addition of `typesContainer`.

pgaskin, 2022-12-08:
If there aren't any objections, I'll do a release in a few hours since this was a relatively trivial update.

jackiew1, 2022-12-08:
@pgaskin 

Agreed. Nothing for me to do as far as I can see.

pgaskin, 2022-12-09:
Urrrgh, GitHub revoking my OAuth tokens seems to have completely broken Drone CI. I'll have to make this release manually...

pgaskin, 2022-12-09:
Released: [`v80`](https://github.com/pgaskin/kobopatch-patches/releases/v80), MR:thread-350805.

pgaskin, 2022-12-09:
<details><summary><b>$ git diff v79:src/versions/4.34.20097 v80:src/versions/4.35.20400</b></summary>

```diff
diff --git a/libadobe.so.yaml/pgaskin.yaml b/libadobe.so.yaml/pgaskin.yaml
index 2533684..a39aa41 100644
--- a/libadobe.so.yaml/pgaskin.yaml
+++ b/libadobe.so.yaml/pgaskin.yaml
@@ -6,5 +6,5 @@ Remove PDF map widget shown during panning:
   - BaseAddress:  {Sym: "N3AdobeReader::showMapWidget()"}
   # tail: N3AdobeReader::updatePanningMap() -> N3AdobeReader::hideMapWidget()
   # TODO: figure out what broke the plt parsing in kobopatch for libadobe in 18730+
-  - ReplaceBytes: {Offset:  80, FindInstBW: 0x1BEDC, ReplaceInstBW: 0x1B1EC}
-  - ReplaceBytes: {Offset: 192, FindInstBW: 0x1BEDC, ReplaceInstBW: 0x1B1EC}
+  - ReplaceBytes: {Offset:  80, FindInstBW: 0x1C134, ReplaceInstBW: 0x1B42C}
+  - ReplaceBytes: {Offset: 192, FindInstBW: 0x1C134, ReplaceInstBW: 0x1B42C}
diff --git a/libnickel.so.1.0.0.yaml/pgaskin.yaml b/libnickel.so.1.0.0.yaml/pgaskin.yaml
index b7a7817..4991690 100644
--- a/libnickel.so.1.0.0.yaml/pgaskin.yaml
+++ b/libnickel.so.1.0.0.yaml/pgaskin.yaml
@@ -13,7 +13,7 @@ Both page turn sides go next:
   - Enabled: no
   - Description: Make both page turn sides go next in the reader.
   # In ReadingView::processTap(TapGesture*) where the page direction is checked:
-  - BaseAddress:  {Sym: "ReadingView::processTap(TapGesture*)", Rel: 560}
+  - BaseAddress:  {Sym: "ReadingView::processTap(TapGesture*)", Rel: 564}
   - ReplaceBytes: {Offset:  0, FindInstBLX: {SymPLT: "ReadingView::leftToRightPageProgressDirection() const"}, CheckOnly: true}
   # nop the branch for checking if left-to-right (so it always runs the swap of back/forward)
   - ReplaceBytes: {Offset:  4, FindH: 18 B9, ReplaceInstNOP: true}
@@ -528,7 +528,7 @@ Customize ComfortLight settings:
   # In an unnamed subroutine two layers into FrontLightPopupController::loadView
   # (find it by going back from QTime::addSecs), the times for the dropdown are
   # generated into a QVector<QPair<QString, QTime>> with a simple loop.
-  - BaseAddress: 0xCDF204 # find the base of the unnamed subroutine with the x-ref to _ZN5QTimeC1Eiiii
+  - BaseAddress: 0xCED04C # find the base of the unnamed subroutine with the x-ref to _ZN5QTimeC1Eiiii
   #
   # Change the initial hour / first bedtime dropdown item passed to the QTime
   # constructor (mov r1, #21):
diff --git a/nickel.yaml/oren64.yaml b/nickel.yaml/oren64.yaml
index daeb1d9..63d7c62 100644
--- a/nickel.yaml/oren64.yaml
+++ b/nickel.yaml/oren64.yaml
@@ -27,18 +27,23 @@ Increase headlines font:
 New home screen subtitle custom font:
   - Enabled: no
   - Description: |
-      Increase subtitle font size, font color black.
-      Subtitle examples are the grey text captions containing:
+      On the Home screen:
+        Change upper label font-size
+        Change subtitle (lower label) font size and/or font colour.
+      Upper label example is the black text caption
+      - 'x% Read' - top row
+      - 'My Books' - middle row
+      Subtitle (lower label) examples are the grey text captions containing:
       - 'x HOURS TO GO' for kepubs (not shown for epubs) - top row
       - 'x BOOKS' under 'My Books' - middle row
       Works best when used with with patch `Increase home screen cover size`
   - FindZlib: "#homeWidgetTopContainer" # qss/GenericHomeWidget.qss
   - ReplaceZlibGroup:
       Replacements:
-        # Change subtitle font colour from mid-grey to black
+        # Change subtitle (lower label) font colour from mid-grey to black
         - {Find: "MetaDataLabel {\n  color: #666666;\n  qproperty-indent: 0;\n}", Replace: "MetaDataLabel {\n  color: #000000;\n  qproperty-indent: 0;\n}"}
         #
-        # Increase subtitle font-size
+        # Change subtitle (lower label) font-size (e.g. time remaining in book)
           # Touch/Mini (Trilogy)
         - Find:    "MetaDataLabel[qApp_deviceIsTrilogy=true] {\n  font-size: 13px;\n}"
           Replace: "MetaDataLabel[qApp_deviceIsTrilogy=true] {\n  font-size: 14px;\n}"
@@ -54,6 +59,23 @@ New home screen subtitle custom font:
[... 23 lines cut ...]
diff --git a/nickel.yaml/pgaskin.yaml b/nickel.yaml/pgaskin.yaml
index b2087bb..431a424 100644
--- a/nickel.yaml/pgaskin.yaml
+++ b/nickel.yaml/pgaskin.yaml
@@ -15,7 +15,7 @@ Remove forgot pin button from lock screen:
       Removes the Forgot PIN -> Sign Out button from the lock screen.
       If this is enabled and you forget your pin, you will need to
       hard reset your Kobo.
-  - BaseAddress: 0x17F23D9 # qss/PinCodeInputDialog.qss
+  - BaseAddress: 0x18034F1 # qss/PinCodeInputDialog.qss
   - ReplaceZlib:
       Find: "#lblForgotPin[qApp_deviceIsDragon=true],\n#lblSignOut[qApp_deviceIsDragon=true] {\n  font-size: 26px;\n}"
       Replace: "#lblForgotPin,#lblSignOut{qproperty-visible:false;}"
@@ -23,7 +23,7 @@ Remove forgot pin button from lock screen:
 Increase size of kepub chapter progress chart:
   - Enabled: no
   - Description: Originally by oren64, rewritten for 4.16.13337 by pgaskin (geek1011).
-  - BaseAddress: 0x17F4EBE # qss/ReadingMenuStatsView.qss
+  - BaseAddress: 0x17FE128 # qss/ReadingMenuStatsView.qss
   - ReplaceZlibGroup:
       Replacements:
       # Top padding is already set to 15px, 25px, 33px, vertical aligned to middle.
```

</details>

closed 2022-12-09 by pgaskin

pgaskin, 2022-12-16:
Kobo has now apparently pulled 4.35... and released 4.34 to the Mini (as the first update in 7 years)... is this a mistake?

[image]

cc @gtalusan

gtalusan, 2022-12-16:
Happy holidays!