pgaskin/kobopatch-patches #115: Default kepub fonts

superjeng1, 2022-12-12:
There are already default font patches for ePub books:
* Default ePub serif font (Amasis)
* Default ePub sans-serif font (Gill Sans)
* Default ePub monospace font

Is it possible to change what the kepub reader uses too? The reader does have serif fonts for Chinese, however, it won't use the CJK serif fonts when the book asks for `font-family: serif;`.

superjeng1, 2022-12-27:
Hi, sorry for the late reply. My Locale is set to Chinese, and when I open a Chinese book that uses `font-family: serif;`, it uses a sans-serif font instead. I don't want to override what font the publisher has set, so I cannot just select another font from the menu. That's why I am trying to find a way to change what Kobo uses for serif and sans-serif.

jackiew1, 2022-12-29:
I have very little knowledge about the Chinese/Japanese fonts and locales but I do know that Kobo treats them as special cases. In the nickel CSS resources there are hundreds of cases where the GUI is forced to use a built-in sans-serif font. For example:
```
#caption[localeName="zh"] {
  font-family: Sans-SerifZH-Simplified, sans-serif;
  font-style: normal;
}
#caption[localeName="zh-HK"] {
  font-family: Sans-SerifZH-Traditional, sans-serif;
  font-style: normal;
}
#caption[localeName="zh-TW"] {
  font-family: Sans-SerifZH-Traditional, sans-serif;
  font-style: normal;
}
```
It doesn't look as if the Kobo GUI allows for a difference between serif and sans-serif for Chinese.

I suspect (_but I'm not certain_) that the situation when reading a kepub will be that generic CSS such as `font-family: serif` or `font-family: sans-serif` will automatically use the same built-in font which is used for the rest of the GUI, namely the Chinese sans-serif font you don't like. For most non-Chinese/Japanese locales  in kepubs, `font-family: serif` displays Georgia and `font-family: sans-serif` displays Avenir.

One thing you could try is to replace the Chinese built-in font that you don't like with one that you prefer. This is at your own risk, of course. You would need to prepare your new font before using the `files:` section of `kobopatch.yaml` to copy the new font over the top of the old built-in font. In addition to making sure the new filename matches the old filename, the preparation requires you to use a font editor to change the internal font name of the new font to exactly match the internal font name of the old font, which I assume is `AR UDJingxihei`. If it works your new font would be used throughout the whole GUI as well as when reading kepubs.

If you don't want to do this then, by default, Kobo offers only 2 options for custom fonts in kepubs:
- All kepub text is displayed in the font you select from the [Aa] menu. Any embedded fonts ignored.
- All custom fonts are embedded in each kepub and you must select `Publisher Default` from the [Aa] menu to see them.

The only other option I know of (which I use myself) is to enable patch `Un-Force user font-family in KePubs` in `libnickel.so.1.0.0.yaml`. This offers 3 alternate ways of making the first option above less aggressive in enforcing the selected font from the [Aa] menu. Using this should allow you to choose your favourite font to be used for the main body text from the [Aa] menu, but still allow embedded fonts to show correctly for parts of the text, e.g. headings . How well it works will vary from kepub to kepub depending on its internal HTML/CSS. If you are prepared to edit a book (CSS and/or HTML) before reading it then you can probably guarantee success.

superjeng1, 2022-12-29:
Thank you for the detailed response.

I currently also uses `Un-Force user font-family in KePubs` to achieve the result I wanted, however I am not certain it will work with every publisher. (What I wanted is the system to actually use the font style that the publisher has set)

For the novel series that I am currently reading, this works fine enough since the body is serif and the highlighted text or chat logs are marked as sans-serif. This publisher achieve this by wrapping the text in a `<span class="gfont"></span>`, and set the class to have the `font-family: sans-serif` property in the CSS file. Because of this, I could just enable that patch and change to a serif font as the publisher intended.

Editing the CSS or HTML file could result in a perfectly formatted book, but it means I have to side-load the books, which I don't really like.

I did change the default sans-serif font to another sans-serif font that I liked in my patch yaml file, and can confirm replacing the font file works. However, not for all fonts that is.

I have theorized that, maybe Kobo is still trying to display Georgia for `font-family: serif` and Avenir for `font-family: sans-serif`. So I tried to swap those for a Chinese font. However, it did not work out. Maybe I used the wrong font name when I edited the font files since I don't really know what their names were, but if this is possible, it would be pretty cool though.

jackiew1, 2022-12-29:
> I have theorized that, maybe Kobo is still trying to display Georgia for `font-family: serif` and Avenir for `font-family: sans-serif`. So I tried to swap those for a Chinese font. However, it did [...]

It hadn't occurred to me that Kobo would try to use Georgia and Avenir in Chinese books. It sounds a bit dumb, doesn't it?

I have also replaced the built-in Georgia and Avenir fonts on my own devices. Georgia is simple enough as its internal font name is the same, `Georgia`. However, the correct internal font name for Avenir is `Avenir Next`. Is that what you tried?

superjeng1, 2022-12-29:
I think I have use `Avenir Next` at some point, but I think it did not work.

What I thought is, Kobo may try to use for example `Avenir Next` and then found that the character is missing and used the default Chinese font instead, which turns out to be a serif font.

It also occurred to me that there maybe some sort of limit on the font file size or maybe it has defined the character set available to the font hard coded else where, but that is all guesses without any sort of proof.

jackiew1, 2022-12-29:
One last thought. Have you tried sideloading a Chinese font to the /fonts subdirectory. Then checking it as a 'Supplemental' fallback font in the [Aa] menu when you're reading a kepub?

From my own experience the fairly new kepub fallback fonts feature seems a bit hit-and-miss, but is worth a try if you haven't already.

If you have tried it and it didn't work ... In firmware 4.35.20400 Kobo added the ability to download OTA on-demand some fallback fonts for Korean, Indian, Arabic (and a few more) users. These are all fonts from the Noto family, presumably because they are free to distribute. Does Noto have anything aimed at Chinese users. Perhaps this would succeed where others didn't.

After that, I'm afraid I'm out of ideas.