pgaskin/kobopatch-patches #119: Patches for 4.26.21095

pgaskin, 2023-04-20:
- [ ] Upload testdata (@pgaskin)
- [ ] Extract resources (@pgaskin)
- [ ] Find libnickel offsets where necessary (@pgaskin)
- [ ] Look at new Elipsa 2E stuff, find device info (@pgaskin, @NiLuJe, @jackiew1, @shermp)
- [ ] Look at new features
- [ ] Update most patches (@pgaskin)
- [ ] Update other patches if necessary (@pgaskin, @shermp)
- [ ] Update nickel patches if necessary (@jackiew1?)
- No release, probably.
- Later (from #112): Add Clara/Elipsa 2E to koboutils (@pgaskin)

pgaskin, 2023-04-20:
Elipsa 2E:
- `condor`
- 389

jackiew1, 2023-04-21:
I've had a quick look at nickel CSS for mark11. There's no specific reference to `qApp_deviceIsCondor` in there so I assume the Elipsa2E will just use whatever settings the Elipsa1 (i.e. Europa, but mainly Daylight) uses.

There are 7 extra zlib CSS streams in nickel but I can't identify the new ones until you've created the usual resources stuff. Bear in mind that it's 01:10 here in the UK, so it probably won't be till tomorrow.

jackiew1, 2023-04-21:
@pgaskin Just noticed that the fw number in this issue's title is wrong.

NiLuJe, 2023-04-21:
Welp, `mt8113t-ntx`.

jackiew1, 2023-04-21:
> Welp, `mt8113t-ntx`.

Is that good ... or bad?

NiLuJe, 2023-04-21:
Amazon did wonderful things with an MTK SoC on the PW5/Scribe. Jury's still out whether NTX can pull something like it ;).

pgaskin, 2023-04-22:
> ETA: I see GDriveFileWidget and GDriveFolderWidget in there for all devices. Were we expecting a new feature involving GoogleDrive?

I believe so; they're doing the same thing as they did with dropbox in libnickel.

pgaskin, 2023-04-22:
> Is that good ... or bad?

At least they appear to be using the same toolchain and target architecture.

pgaskin, 2023-05-09:
Mostly the same as usual, but libadobe offsets need to be updated:

```
Errors:
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.36.21095/nickel.yaml/pgaskin.yaml: Increase size of kepub chapter progress chart: could not apply patch "Increase size of kepub chapter progress chart": line 28: inst 4: ReplaceZlib: not a zlib stream
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.36.21095/nickel.yaml/pgaskin.yaml: Remove forgot pin button from lock screen: could not apply patch "Remove forgot pin button from lock screen": line 20: inst 4: ReplaceZlib: not a zlib stream
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.36.21095/libnickel.so.1.0.0.yaml/pgaskin.yaml: Customize ComfortLight settings: could not apply patch "Customize ComfortLight settings": line 535: inst 4: ReplaceInt: could not find specified bytes at offset
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.36.21095/libnickel.so.1.0.0.yaml/pgaskin.yaml: Hide browser from beta features: could not apply patch "Hide browser from beta features": line 82: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.36.21095/libadobe.so.yaml/pgaskin.yaml: Remove PDF map widget shown during panning: could not apply patch "Remove PDF map widget shown during panning": line 9: inst 4: ReplaceBytes: could not find specified bytes
```

Also, resource extraction is currently broken due to a unicorn arg error when attempting to map the second PT_LOAD segment. The only difference appears to be that in all previous versions the p_filesz was 2608, but in this one, it's 2688. I'll have to look into this more later.

I'll also need to look at how the new device codename lines up with the rest.

jackiew1, 2023-05-09:
@pgaskin, I still don't see this fw available for anything other than the Elipsa2E. Do you know something that I don't or are you planning a release for Elipsa2E only?

pgaskin, 2023-06-23:
**Resource extraction:**

```bash
./qrc2zip --output "nickel.21095.qInitResources_styles.zip"       --recursive --verbose "nickel" 1 26012752 25873456 26001224 || exit 1
./qrc2zip --output "nickel.21095.qInitResources_resources.zip"    --recursive --verbose "nickel" 1 21066912   100192 21065584 || exit 1
./qrc2zip --output "nickel.21095.qInitResources_certificates.zip" --recursive --verbose "nickel" 1 26174136 26015976 26169960 || exit 1
./qrc2zip --output "nickel.21095.qInitResources_translations.zip" --recursive --verbose "nickel" 1 25873128 21067952 25872384 || exit 1
```

<details><summary><b>Output</b></summary>

https://gist.github.com/pgaskin/b30a1dd75dc6a24053efa4c61bac7c00

</details>

<details><summary><b>$ git diff --no-index 20400 21095</b></summary>

https://gist.github.com/pgaskin/2b766061f469d9886c70a980b60daa54

</details>

---

https://krc.storage.pgaskin.net/nickel.21095.qInitResources_certificates.zip
https://krc.storage.pgaskin.net/nickel.21095.qInitResources_resources.zip
https://krc.storage.pgaskin.net/nickel.21095.qInitResources_styles.zip
https://krc.storage.pgaskin.net/nickel.21095.qInitResources_translations.zip

pgaskin, 2023-06-23:
I don't see anything relevant in the resources diff, @jackiew1, so I think this one won't need changes.

pgaskin, 2023-06-23:
Released: [`v81`](https://github.com/pgaskin/kobopatch-patches/releases/v81), MR:thread-354675.

pgaskin, 2023-06-23:
<details><summary><b>$ git diff v80:src/versions/4.35.20400 v81:src/versions/4.36.21095</b></summary>

```diff
diff --git a/libadobe.so.yaml/pgaskin.yaml b/libadobe.so.yaml/pgaskin.yaml
index a39aa41..f892ff8 100644
--- a/libadobe.so.yaml/pgaskin.yaml
+++ b/libadobe.so.yaml/pgaskin.yaml
@@ -6,5 +6,5 @@ Remove PDF map widget shown during panning:
   - BaseAddress:  {Sym: "N3AdobeReader::showMapWidget()"}
   # tail: N3AdobeReader::updatePanningMap() -> N3AdobeReader::hideMapWidget()
   # TODO: figure out what broke the plt parsing in kobopatch for libadobe in 18730+
-  - ReplaceBytes: {Offset:  80, FindInstBW: 0x1C134, ReplaceInstBW: 0x1B42C}
-  - ReplaceBytes: {Offset: 192, FindInstBW: 0x1C134, ReplaceInstBW: 0x1B42C}
+  - ReplaceBytes: {Offset:  80, FindInstBW: 0x1C1DC, ReplaceInstBW: 0x1B4D4}
+  - ReplaceBytes: {Offset: 192, FindInstBW: 0x1C1DC, ReplaceInstBW: 0x1B4D4}
diff --git a/libnickel.so.1.0.0.yaml/pgaskin.yaml b/libnickel.so.1.0.0.yaml/pgaskin.yaml
index 4991690..cdbbf48 100644
--- a/libnickel.so.1.0.0.yaml/pgaskin.yaml
+++ b/libnickel.so.1.0.0.yaml/pgaskin.yaml
@@ -80,7 +80,7 @@ Hide browser from beta features:
   - Description: Hides the built-in browser from beta features.
   - ReplaceBytes:
       Base:        "N3SettingsExtrasView::N3SettingsExtrasView(QWidget*)"
-      Offset:      1392
+      Offset:      1516
       FindInstBLX: {SymPLT: "Device::isParentalControlEnabled() const"}
       ReplaceH:    4F F0 01 00
   - ReplaceBytes:
@@ -528,7 +528,7 @@ Customize ComfortLight settings:
   # In an unnamed subroutine two layers into FrontLightPopupController::loadView
   # (find it by going back from QTime::addSecs), the times for the dropdown are
   # generated into a QVector<QPair<QString, QTime>> with a simple loop.
-  - BaseAddress: 0xCED04C # find the base of the unnamed subroutine with the x-ref to _ZN5QTimeC1Eiiii
+  - BaseAddress: 0xE61B6C # find the base of the unnamed subroutine with the x-ref to _ZN5QTimeC1Eiiii
   #
   # Change the initial hour / first bedtime dropdown item passed to the QTime
   # constructor (mov r1, #21):
diff --git a/nickel.yaml/geoffr.yaml b/nickel.yaml/geoffr.yaml
index a82c377..9c7dec9 100644
--- a/nickel.yaml/geoffr.yaml
+++ b/nickel.yaml/geoffr.yaml
@@ -16,7 +16,7 @@ Reduce top/bottom page spacer:
       # Glo/Aura/Aura2ed: 32px --> 16px
       - Find:    "MediumVertSpacer[qApp_deviceIsPhoenix=true] {\n  min-height: 32px;\n  max-height: 32px;\n}"
         Replace: "MediumVertSpacer[qApp_deviceIsPhoenix=true] {\n  min-height: 16px;\n  max-height: 16px;\n}"
-      # AuraHD/H2O/GloHD/ClaraHD: 44px --> 22px
+      # AuraHD/H2O/GloHD/ClaraHD/Clara2E: 44px --> 22px
       - Find:    "MediumVertSpacer[qApp_deviceIsDragon=true] {\n  min-height: 44px;\n  max-height: 44px;\n}"
         Replace: "MediumVertSpacer[qApp_deviceIsDragon=true] {\n  min-height: 22px;\n  max-height: 22px;\n}"
       # AuraOne/Forma/Sage/Elipsa: 56px --> 28px
@@ -36,7 +36,7 @@ Custom synopsis details line spacing:
       - {Find: "line-height: 1.45em;", Replace: "line-height: 1.3em;"}
         # Glo/Aura/Aura2E (Phoenix)
       - {Find: "line-height: 1.4em;",  Replace: "line-height: 1.3em;"}
-        # AuraHD/AuraH2O/AuraH2O2/GloHD/ClaraHD (Dragon) and
+        # AuraHD/AuraH2O/AuraH2O2/GloHD/ClaraHD/Clara2E (Dragon) and
         # LibraH2O/Libra2 (Storm) and
         # AuraOne/Forma/Sage/Elipsa (Daylight) 
       - {Find: "line-height: 1.35em;", Replace: "line-height: 1.3em;"}
@@ -50,7 +50,7 @@ Custom synopsis font size:
       - {Find: "font-size: 19px;", Replace: "font-size: 20px;"} # Touch/Mini (Trilogy)
       - {Find: "font-size: 23px;", Replace: "font-size: 25px;"} # Glo/Aura/Aura2E/Nia (Phoenix)
       - {Find: "font-size: 29px;", Replace: "font-size: 32px;"} # AuraHD/AuraH2O/AuraH2O2 (Dragon)
-      - {Find: "font-size: 32px;", Replace: "font-size: 35px;"} # GloHD/ClaraHD (Alyssum/Nova)
+      - {Find: "font-size: 32px;", Replace: "font-size: 35px;"} # GloHD/ClaraHD/Clara2E (Alyssum/Nova)
       - {Find: "font-size: 34px;", Replace: "font-size: 37px;"} # LibraH2O/Libra2 (Storm)
       - {Find: "font-size: 37px;", Replace: "font-size: 40px;"} # AuraONE/Forma/Sage/Elipsa (Daylight)
 
@@ -67,7 +67,7 @@ Increase home screen cover size:
         # Glo/Aura/Aura2E/Nia (Phoenix)
[... 186 lines cut ...]
diff --git a/nickel.yaml/pgaskin.yaml b/nickel.yaml/pgaskin.yaml
index 431a424..aff1667 100644
--- a/nickel.yaml/pgaskin.yaml
+++ b/nickel.yaml/pgaskin.yaml
@@ -15,7 +15,7 @@ Remove forgot pin button from lock screen:
       Removes the Forgot PIN -> Sign Out button from the lock screen.
       If this is enabled and you forget your pin, you will need to
       hard reset your Kobo.
-  - BaseAddress: 0x18034F1 # qss/PinCodeInputDialog.qss
+  - BaseAddress: 0x18CA152 # qss/PinCodeInputDialog.qss
   - ReplaceZlib:
       Find: "#lblForgotPin[qApp_deviceIsDragon=true],\n#lblSignOut[qApp_deviceIsDragon=true] {\n  font-size: 26px;\n}"
       Replace: "#lblForgotPin,#lblSignOut{qproperty-visible:false;}"
@@ -23,7 +23,7 @@ Remove forgot pin button from lock screen:
 Increase size of kepub chapter progress chart:
   - Enabled: no
   - Description: Originally by oren64, rewritten for 4.16.13337 by pgaskin (geek1011).
-  - BaseAddress: 0x17FE128 # qss/ReadingMenuStatsView.qss
+  - BaseAddress: 0x18B56F6 # qss/ReadingMenuStatsView.qss
   - ReplaceZlibGroup:
       Replacements:
       # Top padding is already set to 15px, 25px, 33px, vertical aligned to middle.
```

</details>

closed 2023-06-23 by pgaskin