pgaskin/kobopatch-patches #120: Musl library compatibility?

sdothum, 2023-04-29:
Earlier this year i migrated from the glibc version of Void linux to Alpine linux (which is based on the musl library).

i just did a test of running kobopatch (which i used to update my Libra 2 when i was still on Void) and kobopatch.sh now fails (cannot execute bin/kobopatch-linux-64bit.

i am assuming this is due to the musl based distribution i am now using and guessing i need to build the appropriate binaries for kobopatches on this platform.

i've not been looking in the right places.. where do i find the repos for building kobopatch-linux-64bit?

sdothum, 2023-04-29:
Think i've found it at "https://github.com/pgaskin/kobopatch/releases"

NiLuJe, 2023-04-29:
This is pure go, so I'd probably blame PEBCAK instead ;).

In any case, kobopatch is over here: https://github.com/pgaskin/kobopatch

sdothum, 2023-04-29:
Thanks.. i did find that repo but wasn't sure it was what i needed at first glance.

i did manage to create a KoboRoot.tgz after generating just the build/kobopatch-linux-64bit go binary and copying it over to the kobopatches bin/ directory -- after looking at the .drone.yml file (without creating a docker container).

Hopefully, that's it!

closed 2023-04-29 by sdothum

pgaskin, 2023-06-23:
> This is pure go

Not quite; zlib is native and requires CGO. I'm planning to make it a WebAssembly module or switch to native Go after integrating qrc, but for now, it does depend on the host libc.