pgaskin/kobopatch-patches #121: Patches for 4.36.21533

pgaskin, 2023-06-23:
- [ ] Upload testdata (@pgaskin)
- [ ] Fix armqrc (@pgaskin)
- [ ] Extract resources (@pgaskin)
- [ ] Find libnickel offsets where necessary (@pgaskin)
- [ ] Update most patches (@pgaskin)
- [ ] Update other patches if necessary (@pgaskin, @shermp)
- [ ] Update nickel patches if necessary (@jackiew1?)
- Later (from #112): Add Clara/Elipsa 2E to koboutils (@pgaskin)

pgaskin, 2023-06-23:
**Resource extraction:**

```bash
./qrc2zip --output "nickel.21533.qInitResources_styles.zip"       --recursive --verbose "nickel" 1 26572336 26432288 26560696 || exit 1
./qrc2zip --output "nickel.21533.qInitResources_resources.zip"    --recursive --verbose "nickel" 1 21580752   101096 21579424 || exit 1
./qrc2zip --output "nickel.21533.qInitResources_certificates.zip" --recursive --verbose "nickel" 1 26733744 26575584 26729568 || exit 1
./qrc2zip --output "nickel.21533.qInitResources_translations.zip" --recursive --verbose "nickel" 1 26431960 21581792 26431216 || exit 1
```

<details><summary><b>Output</b></summary>

https://gist.github.com/pgaskin/87e57940310071b1ee118c11f34691df

</details>

<details><summary><b>$ git diff --no-index 21095 21533</b></summary>

https://gist.github.com/pgaskin/0cc385838abda7bb026e14b434150728

</details>

---

https://krc.storage.pgaskin.net/nickel.21533.qInitResources_certificates.zip
https://krc.storage.pgaskin.net/nickel.21533.qInitResources_resources.zip
https://krc.storage.pgaskin.net/nickel.21533.qInitResources_styles.zip
https://krc.storage.pgaskin.net/nickel.21533.qInitResources_translations.zip

pgaskin, 2023-06-23:
```
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.37.21533/libnickel.so.1.0.0.yaml/geoffr.yaml: Custom Sleep/Power-off timeouts: could not apply patch "Custom Sleep/Power-off timeouts": line 312: inst 13: ReplaceInt: could not find specified bytes at offset
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.37.21533/nickel.yaml/pgaskin.yaml: Remove forgot pin button from lock screen: could not apply patch "Remove forgot pin button from lock screen": line 20: inst 4: ReplaceZlib: not a zlib stream
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.37.21533/nickel.yaml/pgaskin.yaml: Increase size of kepub chapter progress chart: could not apply patch "Increase size of kepub chapter progress chart": line 28: inst 4: ReplaceZlib: not a zlib stream
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.37.21533/libnickel.so.1.0.0.yaml/pgaskin.yaml: Larger Sleep/Power-off timeouts: could not apply patch "Larger Sleep/Power-off timeouts": line 388: inst 16: ReplaceInt: could not find specified bytes at offset
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.37.21533/libnickel.so.1.0.0.yaml/pgaskin.yaml: Customize ComfortLight settings: could not apply patch "Customize ComfortLight settings": line 535: inst 4: ReplaceInt: could not find specified bytes at offset
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.37.21533/libnickel.so.1.0.0.yaml/pgaskin.yaml: Swap reading header/footer: could not apply patch "Swap reading header/footer": line 483: inst 4: ReplaceBytes: could not find specified bytes at offset
```

Quite a bit more than usual, and this one will need extra testing.

pgaskin, 2023-06-23:
Resource diff doesn't look relevant to @jackiew1, but may be slightly interesting to @shermp wrt the reading header size adjustments.

pgaskin, 2023-06-23:
Released: [`v81`](https://github.com/pgaskin/kobopatch-patches/releases/v81), MR:thread-354676.

pgaskin, 2023-06-23:
<details><summary><b>$ git diff v81:src/versions/4.36.21095 v81:src/versions/4.37.21533</b></summary>

```diff
diff --git a/libnickel.so.1.0.0.yaml/geoffr.yaml b/libnickel.so.1.0.0.yaml/geoffr.yaml
index d997055..d14f644 100644
--- a/libnickel.so.1.0.0.yaml/geoffr.yaml
+++ b/libnickel.so.1.0.0.yaml/geoffr.yaml
@@ -306,22 +306,13 @@ Custom Sleep/Power-off timeouts:
   - ReplaceString: {Offset: 32, Find: "45 mins",  Replace: "2 hours"}
   - ReplaceString: {Offset: 38, Find: "60 mins",  Replace: "4 hours"}
     # Replace values MOVed then added to the list in N3SettingsPowerView::initChoices:
-    # Actual values used for sleep menu
-  - BaseAddress: {Sym: "N3SettingsPowerView::initChoices()", Rel: 12}
-  - ReplaceInt:  {Offset:   0, Find: 5,  Replace:  10}
-  - ReplaceInt:  {Offset:  88, Find: 10, Replace:  20}
-  - ReplaceInt:  {Offset: 148, Find: 15, Replace:  30}
-  - ReplaceInt:  {Offset: 208, Find: 30, Replace:  60}
-  - ReplaceInt:  {Offset: 268, Find: 45, Replace: 120}
-  - ReplaceInt:  {Offset: 328, Find: 60, Replace: 240}
-    # Actual values used for power menu
-  - BaseAddress: {Sym: "N3SettingsPowerView::initChoices()", Rel: 466}
-  - ReplaceInt:  {Offset:   0, Find: 5,  Replace:  10}
-  - ReplaceInt:  {Offset:  60, Find: 10, Replace:  20}
-  - ReplaceInt:  {Offset: 118, Find: 15, Replace:  30}
-  - ReplaceInt:  {Offset: 176, Find: 30, Replace:  60}
-  - ReplaceInt:  {Offset: 234, Find: 45, Replace: 120}
-  - ReplaceInt:  {Offset: 290, Find: 60, Replace: 240}
+  - BaseAddress: "N3SettingsPowerView::initChoices()"
+  - ReplaceInt: {Offset:  56, Find:  5, Replace:   10}
+  - ReplaceInt: {Offset: 118, Find: 10, Replace:   20}
+  - ReplaceInt: {Offset: 180, Find: 15, Replace:   30}
+  - ReplaceInt: {Offset: 240, Find: 30, Replace:   60}
+  - ReplaceInt: {Offset: 300, Find: 45, Replace:  120}
+  - ReplaceInt: {Offset: 360, Find: 60, Replace:  240}
 
 Set KePub hyphenation:
   - Enabled: no
diff --git a/libnickel.so.1.0.0.yaml/pgaskin.yaml b/libnickel.so.1.0.0.yaml/pgaskin.yaml
index cdbbf48..df5cc02 100644
--- a/libnickel.so.1.0.0.yaml/pgaskin.yaml
+++ b/libnickel.so.1.0.0.yaml/pgaskin.yaml
@@ -349,7 +349,7 @@ Larger Sleep/Power-off timeouts:
   - PatchGroup: Sleep/Power-off timeouts
   - Description: |
       Increase the available sleep/power-off timeouts to larger values (up to a
-      few weeks). To customize this patch, see https://pgaskin.net/kobopatch-patches/lgrpwroff-15505.html .
+      few weeks). To customize this patch, see https://pgaskin.net/kobopatch-patches/lgrpwroff-21533.html .
 
       IMPORTANT - Although this patch has been tested by multiple users and
       confirmed to be working, the interaction between the RTC timer and Nickel
@@ -383,19 +383,13 @@ Larger Sleep/Power-off timeouts:
   - ReplaceString: {Offset: 32, Find: "45 mins",  Replace: "12h5m"}
   - ReplaceString: {Offset: 38, Find: "60 mins",  Replace: "24h1m"}
   # Values (see https://www.mobileread.com/forums/showpost.php?p=3887105)
-  - FindBaseAddressSymbol: "N3SettingsPowerView::initChoices()"
-  - ReplaceInt: {Offset:  11, Find:  5, Replace:   1}
-  - ReplaceInt: {Offset: 465, Find:  5, Replace:   1}
-  - ReplaceInt: {Offset:  99, Find: 10, Replace:   4}
-  - ReplaceInt: {Offset: 525, Find: 10, Replace:   4}
-  - ReplaceInt: {Offset: 159, Find: 15, Replace:   7}
-  - ReplaceInt: {Offset: 583, Find: 15, Replace:   7}
-  - ReplaceInt: {Offset: 219, Find: 30, Replace:  28}
-  - ReplaceInt: {Offset: 641, Find: 30, Replace:  28}
-  - ReplaceInt: {Offset: 279, Find: 45, Replace:  83}
-  - ReplaceInt: {Offset: 699, Find: 45, Replace:  83}
-  - ReplaceInt: {Offset: 339, Find: 60, Replace: 165}
-  - ReplaceInt: {Offset: 755, Find: 60, Replace: 165}
+  - BaseAddress: "N3SettingsPowerView::initChoices()"
+  - ReplaceInt: {Offset:  56, Find:  5, Replace:   1}
+  - ReplaceInt: {Offset: 118, Find: 10, Replace:   4}
[... 31 lines cut ...]
diff --git a/nickel.yaml/pgaskin.yaml b/nickel.yaml/pgaskin.yaml
index aff1667..75208ff 100644
--- a/nickel.yaml/pgaskin.yaml
+++ b/nickel.yaml/pgaskin.yaml
@@ -15,7 +15,7 @@ Remove forgot pin button from lock screen:
       Removes the Forgot PIN -> Sign Out button from the lock screen.
       If this is enabled and you forget your pin, you will need to
       hard reset your Kobo.
-  - BaseAddress: 0x18CA152 # qss/PinCodeInputDialog.qss
+  - BaseAddress: 0x1941D42 # qss/PinCodeInputDialog.qss
   - ReplaceZlib:
       Find: "#lblForgotPin[qApp_deviceIsDragon=true],\n#lblSignOut[qApp_deviceIsDragon=true] {\n  font-size: 26px;\n}"
       Replace: "#lblForgotPin,#lblSignOut{qproperty-visible:false;}"
@@ -23,7 +23,7 @@ Remove forgot pin button from lock screen:
 Increase size of kepub chapter progress chart:
   - Enabled: no
   - Description: Originally by oren64, rewritten for 4.16.13337 by pgaskin (geek1011).
-  - BaseAddress: 0x18B56F6 # qss/ReadingMenuStatsView.qss
+  - BaseAddress: 0x194BE97 # qss/ReadingMenuStatsView.qss
   - ReplaceZlibGroup:
       Replacements:
       # Top padding is already set to 15px, 25px, 33px, vertical aligned to middle.
```

</details>

jackiew1, 2023-06-23:
@pgaskin 

Sorry I wasn't around last night. This release looks a bit of a non-event for those without note-taking devices. All my patches should work as-is.

For completeness all the comments in nickel.yaml need to add Elipsa2E to wherever Elipsa is currently found. I can deal with this later today, but obviously too late for your earlier release.

P.S. The fw version is wrong in this heading!

pgaskin, 2023-06-25:
@jackiew1 

Thanks! I've fixed the issue title, and I'll release your changes as soon as there's either a new patch or a firmware release.

One hint for future modifications: you might find it easier to make changes via https://github.dev/pgaskin/kobopatch-patches).

closed 2023-06-25 by pgaskin

jackiew1, 2023-06-25:
> One hint for future modifications: you might find it easier to make changes via [GitHub.dev](https://github.dev/pgaskin/kobopatch-patches), especially when they involve multiple files.

My knowledge of GitHub is very poor and I don't have any real interest in spending time learning more as it's not something I want to use for my own projects. If I could do damage by using [GitHub.dev] wrongly it's probably best if I stay away from it.