pgaskin/kobopatch-patches #123: Patches for 4.37.21582

pgaskin, 2023-07-05:
- [ ] Upload testdata (@pgaskin)
- [ ] Extract resources (@pgaskin)
- [ ] Find libnickel offsets where necessary (@pgaskin)
- [ ] Update most patches (@pgaskin)
- [ ] Test patches (@pgaskin)
- [ ] Make release (@pgaskin)
- [ ] MR post (@pgaskin)
- Later (from #112): Add Clara/Elipsa 2E to koboutils (@pgaskin)
- #122

pgaskin, 2023-07-05:
**Resource extraction:**

```bash
go run github.com/pgaskin/qrc/cmd/qrc2zip@latest --output nickel.21582.qInitResources_styles.zip --recursive --verbose nickel 1 26572352 26432304 26560712
go run github.com/pgaskin/qrc/cmd/qrc2zip@latest --output nickel.21582.qInitResources_resources.zip --recursive --verbose nickel 1 21580752 101096 21579424
go run github.com/pgaskin/qrc/cmd/qrc2zip@latest --output nickel.21582.qInitResources_certificates.zip --recursive --verbose nickel 1 26733760 26575600 26729584
go run github.com/pgaskin/qrc/cmd/qrc2zip@latest --output nickel.21582.qInitResources_translations.zip --recursive --verbose nickel 1 26431976 21581792 26431232
```

[Output](https://gist.github.com/pgaskin/6c33ff60447ff61721e4dbe79fd6389e) [Diff](https://gist.github.com/pgaskin/f33f762976e8f7b23877bd742663f24e)

---

https://krc.storage.pgaskin.net/nickel.21582.qInitResources_certificates.zip
https://krc.storage.pgaskin.net/nickel.21582.qInitResources_resources.zip
https://krc.storage.pgaskin.net/nickel.21582.qInitResources_styles.zip
https://krc.storage.pgaskin.net/nickel.21582.qInitResources_translations.zip

pgaskin, 2023-07-05:
```
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.37.21582/libnickel.so.1.0.0.yaml/pgaskin.yaml: Customize ComfortLight settings: could not apply patch "Customize ComfortLight settings": line 529: inst 4: ReplaceInt: could not find specified bytes at offset
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.37.21582/nickel.yaml/pgaskin.yaml: Increase size of kepub chapter progress chart: could not apply patch "Increase size of kepub chapter progress chart": line 28: inst 4: ReplaceZlib: not a zlib stream
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.37.21582/nickel.yaml/pgaskin.yaml: Remove forgot pin button from lock screen: could not apply patch "Remove forgot pin button from lock screen": line 20: inst 4: ReplaceZlib: not a zlib stream
```

The easiest release possible.

pgaskin, 2023-07-06:
Released: [`v82`](https://github.com/pgaskin/kobopatch-patches/releases/v82), MR:thread-354898.

pgaskin, 2023-07-06:
<details><summary><b>$ git diff v81:src/versions/4.37.21533 v82:src/versions/4.37.21582</b></summary>

```diff
diff --git a/libnickel.so.1.0.0.yaml/pgaskin.yaml b/libnickel.so.1.0.0.yaml/pgaskin.yaml
index df5cc02..5a6ea98 100644
--- a/libnickel.so.1.0.0.yaml/pgaskin.yaml
+++ b/libnickel.so.1.0.0.yaml/pgaskin.yaml
@@ -522,7 +522,7 @@ Customize ComfortLight settings:
   # In an unnamed subroutine two layers into FrontLightPopupController::loadView
   # (find it by going back from QTime::addSecs), the times for the dropdown are
   # generated into a QVector<QPair<QString, QTime>> with a simple loop.
-  - BaseAddress: 0xE7D2D4 # find the base of the unnamed subroutine with the x-ref to _ZN5QTimeC1Eiiii
+  - BaseAddress: 0xE7D454 # find the base of the unnamed subroutine with the x-ref to _ZN5QTimeC1Eiiii
   #
   # Change the initial hour / first bedtime dropdown item passed to the QTime
   # constructor (mov r1, #21):
diff --git a/nickel.yaml/geoffr.yaml b/nickel.yaml/geoffr.yaml
index 9c7dec9..2d48dcd 100644
--- a/nickel.yaml/geoffr.yaml
+++ b/nickel.yaml/geoffr.yaml
@@ -19,7 +19,7 @@ Reduce top/bottom page spacer:
       # AuraHD/H2O/GloHD/ClaraHD/Clara2E: 44px --> 22px
       - Find:    "MediumVertSpacer[qApp_deviceIsDragon=true] {\n  min-height: 44px;\n  max-height: 44px;\n}"
         Replace: "MediumVertSpacer[qApp_deviceIsDragon=true] {\n  min-height: 22px;\n  max-height: 22px;\n}"
-      # AuraOne/Forma/Sage/Elipsa: 56px --> 28px
+      # AuraOne/Forma/Sage/Elipsa/Elipsa2E: 56px --> 28px
       - Find:    "MediumVertSpacer[qApp_deviceIsDaylight=true] {\n  min-height: 56px;\n  max-height: 56px;\n}"
         Replace: "MediumVertSpacer[qApp_deviceIsDaylight=true] {\n  min-height: 28px;\n  max-height: 28px;\n}"
       # LibraH2O/Libra2: 50px --> 25px
@@ -38,7 +38,7 @@ Custom synopsis details line spacing:
       - {Find: "line-height: 1.4em;",  Replace: "line-height: 1.3em;"}
         # AuraHD/AuraH2O/AuraH2O2/GloHD/ClaraHD/Clara2E (Dragon) and
         # LibraH2O/Libra2 (Storm) and
-        # AuraOne/Forma/Sage/Elipsa (Daylight) 
+        # AuraOne/Forma/Sage/Elipsa/Elipsa2E (Daylight) 
       - {Find: "line-height: 1.35em;", Replace: "line-height: 1.3em;"}
 
 Custom synopsis font size:
@@ -52,7 +52,7 @@ Custom synopsis font size:
       - {Find: "font-size: 29px;", Replace: "font-size: 32px;"} # AuraHD/AuraH2O/AuraH2O2 (Dragon)
       - {Find: "font-size: 32px;", Replace: "font-size: 35px;"} # GloHD/ClaraHD/Clara2E (Alyssum/Nova)
       - {Find: "font-size: 34px;", Replace: "font-size: 37px;"} # LibraH2O/Libra2 (Storm)
-      - {Find: "font-size: 37px;", Replace: "font-size: 40px;"} # AuraONE/Forma/Sage/Elipsa (Daylight)
+      - {Find: "font-size: 37px;", Replace: "font-size: 40px;"} # AuraONE/Forma/Sage/Elipsa/Elipsa2E (Daylight)
 
 Increase home screen cover size:
   - Enabled: no
@@ -70,7 +70,7 @@ Increase home screen cover size:
         # AuraHD/AuraH2O/AuraH2O2/GloHD/ClaraHD/Clara2E (Dragon)
       - {Find: "qproperty-leftMargin: 57px;",  Replace: "qproperty-leftMargin: 22px;"}
       - {Find: "qproperty-rightMargin: 57px;", Replace: "qproperty-rightMargin: 22px;"}
-        # AuraONE/Forma/Sage/Elipsa (Daylight)
+        # AuraONE/Forma/Sage/Elipsa/Elipsa2E (Daylight)
       - {Find: "qproperty-leftMargin: 74px;",  Replace: "qproperty-leftMargin: 29px;"}
       - {Find: "qproperty-rightMargin: 74px;", Replace: "qproperty-rightMargin: 29px;"}
         # LibraH2O/Libra2 (Storm)
diff --git a/nickel.yaml/jackie_w.yaml b/nickel.yaml/jackie_w.yaml
index e2c88ef..ddc81e3 100644
--- a/nickel.yaml/jackie_w.yaml
+++ b/nickel.yaml/jackie_w.yaml
@@ -23,7 +23,7 @@ Dictionary pop-up - increase available text area:
           # AuraHD/AuraH2O/AuraH2O2/GloHD/ClaraHD/Clara2E/LibraH2O/Libra2
         - Find:    "#dictionary[qApp_deviceIsDragon=true] {\n  margin-top: 30px;\n  margin-left: 30px;\n}"
           Replace: "#dictionary[qApp_deviceIsDragon=true] {\n  margin-top: 10px;\n  margin-left: 0px;\n}"
-          # AuraOne/Forma/Sage/Elipsa
+          # AuraOne/Forma/Sage/Elipsa/Elipsa2E
         - Find:    "#dictionary[qApp_deviceIsDaylight=true] {\n  margin-top: 42px;\n  margin-left: 42px;\n}"
           Replace: "#dictionary[qApp_deviceIsDaylight=true] {\n  margin-top: 14px;\n  margin-left: 0px;\n}"
         #
@@ -37,7 +37,7 @@ Dictionary pop-up - increase available text area:
[... 163 lines cut ...]
diff --git a/nickel.yaml/pgaskin.yaml b/nickel.yaml/pgaskin.yaml
index 75208ff..e45b4ea 100644
--- a/nickel.yaml/pgaskin.yaml
+++ b/nickel.yaml/pgaskin.yaml
@@ -15,7 +15,7 @@ Remove forgot pin button from lock screen:
       Removes the Forgot PIN -> Sign Out button from the lock screen.
       If this is enabled and you forget your pin, you will need to
       hard reset your Kobo.
-  - BaseAddress: 0x1941D42 # qss/PinCodeInputDialog.qss
+  - BaseAddress: 0x1941D52 # qss/PinCodeInputDialog.qss
   - ReplaceZlib:
       Find: "#lblForgotPin[qApp_deviceIsDragon=true],\n#lblSignOut[qApp_deviceIsDragon=true] {\n  font-size: 26px;\n}"
       Replace: "#lblForgotPin,#lblSignOut{qproperty-visible:false;}"
@@ -23,7 +23,7 @@ Remove forgot pin button from lock screen:
 Increase size of kepub chapter progress chart:
   - Enabled: no
   - Description: Originally by oren64, rewritten for 4.16.13337 by pgaskin (geek1011).
-  - BaseAddress: 0x194BE97 # qss/ReadingMenuStatsView.qss
+  - BaseAddress: 0x194BEA7 # qss/ReadingMenuStatsView.qss
   - ReplaceZlibGroup:
       Replacements:
       # Top padding is already set to 15px, 25px, 33px, vertical aligned to middle.
```

</details>

closed 2023-07-06 by pgaskin

jackiew1, 2023-07-06:
> The easiest release possible.

Excellent😃 

On a non-patch topic, do you know whether it will be possible to access the new GoogleDrive feature from NickelMenu in the same way you can with Dropbox?

pgaskin, 2023-07-06:
> On a non-patch topic, do you know whether it will be possible to access the new GoogleDrive feature from NickelMenu in the same way you can with Dropbox?

Yeah, that's possible (as is notebooks and the other recent features); I just haven't got around to doing it myself yet.

jackiew1, 2023-07-06:
> Yeah, that's possible (as is notebooks and the other recent features); I just haven't got around to doing it myself yet.

Sounds promising, no rush. I suppose it's a bit of a dilemma as to whether to publish them publicly, though.  I don't use notebooks myself.