pgaskin/kobopatch-patches #127: Patches for 4.38.21908

pgaskin, 2023-09-07:
- [ ] Upload testdata (@pgaskin)
- [ ] Extract resources (@pgaskin)
- [ ] Find libnickel offsets where necessary (@pgaskin)
- [ ] Update most patches (@pgaskin)
- [ ] Update other patches if necessary (@pgaskin, @shermp)
- [ ] Update nickel patches if necessary (@jackiew1?)
- [ ] Test patches (@pgaskin)
- [ ] Make release (@pgaskin)
- [ ] MR post (@pgaskin)
- Ongoing (from #112): pgaskin/koboutils#8 (@NiLuJe)
- #122

pgaskin, 2023-09-07:
**Resource extraction:**

```bash
go run github.com/pgaskin/qrc/cmd/qrc2zip@latest --output nickel.21908.qInitResources_styles.zip --recursive --verbose nickel 1 26762560 26622584 26750920
go run github.com/pgaskin/qrc/cmd/qrc2zip@latest --output nickel.21908.qInitResources_resources.zip --recursive --verbose nickel 1 21737592 101096 21736112
go run github.com/pgaskin/qrc/cmd/qrc2zip@latest --output nickel.21908.qInitResources_certificates.zip --recursive --verbose nickel 1 26923968 26765808 26919792
go run github.com/pgaskin/qrc/cmd/qrc2zip@latest --output nickel.21908.qInitResources_translations.zip --recursive --verbose nickel 1 26622256 21738816 26621512
```

[Output](https://gist.github.com/pgaskin/d88fa2320cfdd1255b40175890a8df59) [Diff](https://gist.github.com/pgaskin/8164ff776fd133ffc9c819840baf1798)

---

https://krc.storage.pgaskin.net/nickel.21908.qInitResources_certificates.zip
https://krc.storage.pgaskin.net/nickel.21908.qInitResources_resources.zip
https://krc.storage.pgaskin.net/nickel.21908.qInitResources_styles.zip
https://krc.storage.pgaskin.net/nickel.21908.qInitResources_translations.zip

pgaskin, 2023-09-07:
```
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.38.21908/libnickel.so.1.0.0.yaml/geoffr.yaml: My 24 line spacing values: could not apply patch "My 24 line spacing values": line 54: inst 9: ReplaceBytes: could not find specified bytes at offset
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.38.21908/libnickel.so.1.0.0.yaml/jackie_w.yaml: Dictionary text font-family/font-size/line-height: could not apply patch "Dictionary text font-family/font-size/line-height": line 53: inst 3: FindReplaceString: FindBaseAddressString: FindBaseAddress: could not find bytes
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.38.21908/nickel.yaml/pgaskin.yaml: Increase size of kepub chapter progress chart: could not apply patch "Increase size of kepub chapter progress chart": line 28: inst 4: ReplaceZlib: not a zlib stream
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.38.21908/nickel.yaml/pgaskin.yaml: Remove forgot pin button from lock screen: could not apply patch "Remove forgot pin button from lock screen": line 20: inst 4: ReplaceZlib: not a zlib stream
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.38.21908/libnickel.so.1.0.0.yaml/pgaskin.yaml: Customize ComfortLight settings: could not apply patch "Customize ComfortLight settings": line 529: inst 4: ReplaceInt: could not find specified bytes at offset
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.38.21908/libnickel.so.1.0.0.yaml/pgaskin.yaml: Hide browser from beta features: could not apply patch "Hide browser from beta features": line 82: inst 3: ReplaceBytes: could not find specified bytes
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.38.21908/libnickel.so.1.0.0.yaml/pgaskin.yaml: Never show Kobo Plus, wishlist, and points SmartLinks: could not apply patch "Never show Kobo Plus, wishlist, and points SmartLinks": line 288: inst 5: ReplaceBytes: could not find specified bytes at offset
  /home/patrick/kp/src/kobopatch-patches/src/versions/4.38.21908/libadobe.so.yaml/pgaskin.yaml: Remove PDF map widget shown during panning: could not apply patch "Remove PDF map widget shown during panning": line 9: inst 4: ReplaceBytes: could not find specified bytes
```

A lot more than usual, and this one will be a bit more work since I'm going to need to verify all of the SmartLinks again, update and test the libadobe offsets, and at least partially rewrite "My 24 line spacing values". I'll work on this when I have more time, probably tomorrow.

@jackiew1 "Dictionary text font-family/font-size/line-height"

jackiew1, 2023-09-07:
@pgaskin 
When you create the MR thread for release of these patches I'll need to add a post to remind users that they'll need to make a fresh copy if they want to customise patch `Dictionary text font-family/font-size/line-height`

jackiew1, 2023-09-07:
... also, forgot to say no changes required for nickel patches as far as I can tell.

pgaskin, 2023-09-08:
> I'll need to add a post

I'll post it around the time of your comment then, with some notice beforehand.

pgaskin, 2023-09-11:
Looks like the 24 line spacing patch fix will be relatively trivial; a nop just made its way into the middle for alignment, so I'll need to recompute all the hardcoded offsets.

jackiew1, 2023-09-11:
> Looks like the 24 line spacing patch fix will be relatively trivial; a nop just made its way into the middle for alignment, so I'll need to recompute all the hardcoded offsets.

Tested, seems fine. Sent you a MR PM.

pgaskin, 2023-09-11:
Released: [`v84`](https://github.com/pgaskin/kobopatch-patches/releases/v84), MR:thread-356106.

pgaskin, 2023-09-11:
<details><summary><b>$ git diff v83:src/versions/4.37.21586 v84:src/versions/4.38.21908</b></summary>

```diff
diff --git a/libadobe.so.yaml/pgaskin.yaml b/libadobe.so.yaml/pgaskin.yaml
index f892ff8..14ee20e 100644
--- a/libadobe.so.yaml/pgaskin.yaml
+++ b/libadobe.so.yaml/pgaskin.yaml
@@ -6,5 +6,5 @@ Remove PDF map widget shown during panning:
   - BaseAddress:  {Sym: "N3AdobeReader::showMapWidget()"}
   # tail: N3AdobeReader::updatePanningMap() -> N3AdobeReader::hideMapWidget()
   # TODO: figure out what broke the plt parsing in kobopatch for libadobe in 18730+
-  - ReplaceBytes: {Offset:  80, FindInstBW: 0x1C1DC, ReplaceInstBW: 0x1B4D4}
-  - ReplaceBytes: {Offset: 192, FindInstBW: 0x1C1DC, ReplaceInstBW: 0x1B4D4}
+  - ReplaceBytes: {Offset:  80, FindInstBW: 0x1C224, ReplaceInstBW: 0x1B51C}
+  - ReplaceBytes: {Offset: 192, FindInstBW: 0x1C224, ReplaceInstBW: 0x1B51C}
diff --git a/libnickel.so.1.0.0.yaml/geoffr.yaml b/libnickel.so.1.0.0.yaml/geoffr.yaml
index d14f644..4ea5a07 100644
--- a/libnickel.so.1.0.0.yaml/geoffr.yaml
+++ b/libnickel.so.1.0.0.yaml/geoffr.yaml
@@ -51,7 +51,7 @@ My 24 line spacing values:
   - ReplaceBytes: {Offset: 20, FindInstBLX: {SymPLT: "QList<double>::append(double const&)"}, ReplaceH: 00 00 00 00} # clear out the BLX
   - ReplaceBytes: {Offset:  0, FindH: 00 00, ReplaceH: A9 46} # mov  sb, r5
   - ReplaceBytes: {Offset:  2, FindH: 00 00, ReplaceH: 00 25} # mov  r5, #0
-  - ReplaceBytes: {Offset:  4, FindH: 6C A3, ReplaceH: ED 00} # lsls r5, r5, #3
+  - ReplaceBytes: {Offset:  4, FindH: 6D A3, ReplaceH: ED 00} # lsls r5, r5, #3
     # loop to load 14 vals from new table
   - ReplaceBytes: {Offset:  6, FindH: D3 E9, ReplaceH: 20 46} # mov  r0, r4
   - ReplaceBytes: {Offset:  8, FindH: 00 23, ReplaceH: 04 A1} # adr  r1, #0x10
@@ -62,7 +62,7 @@ My 24 line spacing values:
   - ReplaceBytes: {Offset: 20, FindH: 00 00, ReplaceH: F7 D1} # bne  #-10
     # jump to the rest of the new code
   - ReplaceBytes: {Offset: 22, FindH: 00 00, ReplaceH: 7B E0} # b #250 (356 − (84 + 22))
-  - ReplaceBytes: {Offset: 24, FindH: 69 A3 D3 E9, ReplaceInstNOP: true} # unused
+  - ReplaceBytes: {Offset: 24, FindH: 6A A3 D3 E9, ReplaceInstNOP: true} # unused
     # new table: 14 8-aligned doubles
   - BaseAddress:  {Sym: "ReadingSettings::lineHeightScalars() const", Rel: 112} # right after previous instruction (84 + 24 + 4)
   - ReplaceBytes: {Offset:  12, FindInstBLX: {SymPLT: "QList<double>::append(double const&)"}, ReplaceH: 00 00 00 00} # first, clear out the BLXs in the way
@@ -70,20 +70,20 @@ My 24 line spacing values:
   - ReplaceBytes: {Offset:  58, FindInstBLX: {SymPLT: "QList<double>::append(double const&)"}, ReplaceH: 00 00 00 00}
   - ReplaceBytes: {Offset:  78, FindInstBLX: {SymPLT: "QList<double>::append(double const&)"}, ReplaceH: 00 00 00 00}
   - ReplaceBytes: {Offset: 100, FindInstBLX: {SymPLT: "QList<double>::append(double const&)"}, ReplaceH: 00 00 00 00}
-  - ReplaceBytes: {Offset:   0, FindH: 00 23 07 F1 80 01 20 46, ReplaceH: 00 00 00 00 00 00 F0 3F} # table of 1.0f values, to be replaced later
+  - ReplaceBytes: {Offset:   0, FindH: 00 23 07 f1 80 01 20 46, ReplaceH: 00 00 00 00 00 00 F0 3F} # table of 1.0f values, to be replaced later
   - ReplaceBytes: {Offset:   8, FindH: 61 E9 1A 23 00 00 00 00, ReplaceH: 00 00 00 00 00 00 F0 3F}
-  - ReplaceBytes: {Offset:  16, FindH: 66 A3 D3 E9 00 23 07 F1, ReplaceH: 00 00 00 00 00 00 F0 3F}
+  - ReplaceBytes: {Offset:  16, FindH: 67 A3 D3 E9 00 23 07 F1, ReplaceH: 00 00 00 00 00 00 F0 3F}
   - ReplaceBytes: {Offset:  24, FindH: 80 01 20 46 61 E9 18 23, ReplaceH: 00 00 00 00 00 00 F0 3F}
   - ReplaceBytes: {Offset:  32, FindH: 00 00 00 00 43 F2 33 33, ReplaceH: 00 00 00 00 00 00 F0 3F}
   - ReplaceBytes: {Offset:  40, FindH: 4F F0 33 32 C3 F6 F3 73, ReplaceH: 00 00 00 00 00 00 F0 3F}
   - ReplaceBytes: {Offset:  48, FindH: 07 F1 80 01 20 46 61 E9, ReplaceH: 00 00 00 00 00 00 F0 3F}
-  - ReplaceBytes: {Offset:  56, FindH: 16 23 00 00 00 00 5D A3, ReplaceH: 00 00 00 00 00 00 F0 3F}
+  - ReplaceBytes: {Offset:  56, FindH: 16 23 00 00 00 00 5E A3, ReplaceH: 00 00 00 00 00 00 F0 3F}
   - ReplaceBytes: {Offset:  64, FindH: D3 E9 00 23 07 F1 80 01, ReplaceH: 00 00 00 00 00 00 F0 3F}
   - ReplaceBytes: {Offset:  72, FindH: 20 46 61 E9 14 23 00 00, ReplaceH: 00 00 00 00 00 00 F0 3F}
   - ReplaceBytes: {Offset:  80, FindH: 00 00 00 23 00 22 C3 F6, ReplaceH: 00 00 00 00 00 00 F0 3F}
   - ReplaceBytes: {Offset:  88, FindH: F8 73 07 F1 80 01 20 46, ReplaceH: 00 00 00 00 00 00 F0 3F}
   - ReplaceBytes: {Offset:  96, FindH: 61 E9 12 23 00 00 00 00, ReplaceH: 00 00 00 00 00 00 F0 3F}
-  - ReplaceBytes: {Offset: 104, FindH: 54 A3 D3 E9 00 23 07 F1, ReplaceH: 00 00 00 00 00 00 F0 3F}
+  - ReplaceBytes: {Offset: 104, FindH: 55 A3 D3 E9 00 23 07 F1, ReplaceH: 00 00 00 00 00 00 F0 3F}
     # jump target after table
   - BaseAddress:  {Sym: "ReadingSettings::lineHeightScalars() const", Rel: 356} # starting right after the second-last QList<double>::append(double const&)
   - ReplaceBytes: {Offset: 18, FindInstBLX: {SymPLT: "QList<double>::append(double const&)"}, ReplaceH: 00 00 00 00} # clear the BLX in the way of the last two replacements
@@ -91,7 +91,7 @@ My 24 line spacing values:
   - ReplaceBytes: {Offset:  0, FindH: 00 23,             ReplaceH: 00 25}       # movs  r5, #0
     # loop to load 10 vals from old table (CDCC CCCC CCCC F03F 1F85)
   - ReplaceBytes: {Offset:  2, FindH: 00 22,             ReplaceH: 20 46}       # mov   r0, r4
[... cut ...]
       Replace: "#lblForgotPin,#lblSignOut{qproperty-visible:false;}"
@@ -23,7 +23,7 @@ Remove forgot pin button from lock screen:
 Increase size of kepub chapter progress chart:
   - Enabled: no
   - Description: Originally by oren64, rewritten for 4.16.13337 by pgaskin (geek1011).
-  - BaseAddress: 0x194EE6F # qss/ReadingMenuStatsView.qss
+  - BaseAddress: 0x196E0F6 # qss/ReadingMenuStatsView.qss
   - ReplaceZlibGroup:
       Replacements:
       # Top padding is already set to 15px, 25px, 33px, vertical aligned to middle.
```

</details>

closed 2023-09-11 by pgaskin