pgaskin/kobopatch-patches #130: Patches for 5.0.175773

pgaskin, 2024-05-08:
This one might be tolino-specific, or future Kobo releases may also be 5.x.

TODO

pgaskin, 2024-05-08:
Okay, so I've managed to get ahold of the firmware files... the API checks that the original version is >5.0.111111.

https://ereaderfiles.kobo.com/firmwares/kobo12/May2024/tolino-qt6-recovery-5.0.175773.zip

Also note that Tolino serial numbers start with T rather than N, but that's irrelevant.

I'll need to update kfwproxy to track the 5.x updates separately.

pgaskin, 2024-05-08:
Some random preliminary notes:

- For other people reading this, don't play with this update package if you don't know what you're doing... you don't want to mess up bl2 (or even uboot at this point, since we don't know how to recover it).
- Qt6? "legacy qt5"?
- Recovery mechanism is completely different.
- Update package seems to resemble what I remember of the old tolino format, will need to look into this more.
- Is this being served for kobo11?
- Rootfs is zstd-compressed in the zstd-compressed recoveryfs.

benhaotang, 2024-05-08:
from the tolino shine color that I just bought yesterday, this is the 0-day update that I got when setting up the device. In the webbrowser the UA determine itself as qt 6.6.5

pgaskin, 2024-05-08:
They added the tolino equivalents of the Clara HD and Forma.

| codename | family | id | name |
| --- | --- | --- | --- |
| loki | dragon | 00000000-0000-0000-0000-000000000676 | tolino Shine 3 |
| freya | dragon | 00000000-0000-0000-0000-000000000677 | tolino Epos 2 |

NiLuJe, 2024-05-08:
> * They're now shipping a modernish kernel... if they do end up porting it to other devices, it'll make it nicer to do stuff like dual-boot mods and so on.

That's just the MTK kernel, it's the exact same build as found on 4.39. (And the same base branch as used by other MTK implementations, e.g., lab126).

pgaskin, 2024-05-09:
Wrote a tool to extract the basic device info from the firmware: https://github.com/pgaskin/koboutils/commit/cea683a4efe3044e1a5b1603e3320ef15f3772e5

| ID                                   | Codename             | Family               | Name                       |
| ---                                  | ---                  | ---                  | ---                        |
| 00000000-0000-0000-0000-000000000310 | trilogy              | trilogy              | Kobo Touch                 |
| 00000000-0000-0000-0000-000000000320 | trilogy              | trilogy              | Kobo Touch                 |
| 00000000-0000-0000-0000-000000000330 | kraken               | phoenix              | Kobo Glo                   |
| 00000000-0000-0000-0000-000000000340 | pixie                | trilogy              | Kobo Mini                  |
| 00000000-0000-0000-0000-000000000350 | dragon               | dragon               | Kobo Aura HD               |
| 00000000-0000-0000-0000-000000000360 | phoenix              | phoenix              | Kobo Aura                  |
| 00000000-0000-0000-0000-000000000370 | dahlia               | dragon               | Kobo Aura H2O              |
| 00000000-0000-0000-0000-000000000371 | alyssum              | dragon               | Kobo Glo HD                |
| 00000000-0000-0000-0000-000000000372 | pika                 | trilogy              | Kobo Touch 2.0             |
| 00000000-0000-0000-0000-000000000373 | daylight             | daylight             | Kobo Aura ONE              |
| 00000000-0000-0000-0000-000000000374 | snow                 | dragon               | Kobo Aura H2O Edition 2    |
| 00000000-0000-0000-0000-000000000375 | star                 | phoenix              | Kobo Aura                  |
| 00000000-0000-0000-0000-000000000376 | nova                 | dragon               | Kobo Clara HD              |
| 00000000-0000-0000-0000-000000000377 | frost                | daylight             | Kobo Forma                 |
| 00000000-0000-0000-0000-000000000378 | snow                 | dragon               | Kobo Aura H2O Edition 2    |
| 00000000-0000-0000-0000-000000000379 | star                 | phoenix              | Kobo Aura                  |
| 00000000-0000-0000-0000-000000000380 | frost                | daylight             | Kobo Forma                 |
| 00000000-0000-0000-0000-000000000381 | daylight             | daylight             | Kobo Aura ONE              |
| 00000000-0000-0000-0000-000000000382 | luna                 | phoenix              | Kobo Nia                   |
| 00000000-0000-0000-0000-000000000383 | cadmus               | daylight             | Kobo Sage                  |
| 00000000-0000-0000-0000-000000000384 | storm                | dragon               | Kobo Libra H2O             |
| 00000000-0000-0000-0000-000000000386 | goldfinch            | dragon               | Kobo Clara 2E              |
| 00000000-0000-0000-0000-000000000387 | europa               | dragon               | Kobo Elipsa                |
| 00000000-0000-0000-0000-000000000388 | io                   | dragon               | Kobo Libra 2               |
| 00000000-0000-0000-0000-000000000389 | condor               | dragon               | Kobo Elipsa 2E             |
| 00000000-0000-0000-0000-000000000676 | loki                 | dragon               | tolino Shine 3             |
| 00000000-0000-0000-0000-000000000677 | freya                | daylight             | tolino Epos 2              |
| 00000000-0000-0000-0000-000000000390 | monzaKobo            | dragon               | Kobo Libra Colour          |
| 00000000-0000-0000-0000-000000000690 | monzaTolino          | dragon               | tolino vision color        |
| 00000000-0000-0000-0000-000000000391 | spaKoboBW            | dragon               | Kobo Clara BW              |
| 00000000-0000-0000-0000-000000000393 | spaKoboColour        | dragon               | Kobo Clara Colour          |
| 00000000-0000-0000-0000-000000000691 | spaTolinoBW          | dragon               | tolino shine 5             |
| 00000000-0000-0000-0000-000000000693 | spaTolinoColour      | dragon               | tolino shine color         |

pgaskin, 2024-05-09:
Also wrote a tool to extract firmware version info from the current kinds of update packages: https://github.com/pgaskin/koboutils/commit/633a819dcde081facdb81e63e96e272ca8a12586

```
$ kobo-fwinfo kobo6-update-4.22.15190.zip kobo11-update-* tolino-qt6-recovery-5.0.175773.zip
kobo6-update-4.22.15190.zip: KoboRoot.tgz package { version=4.22.15190 date=2020-06-22 revision=efcb227b40228eba2d7cd9565e892f19332faf16 }
kobo11-update-4.36.21095.zip: KoboRoot.tgz package { version=4.36.21095 date=2023-04-14 revision=c59487fff360719e3b65b3171e88cadf7e77b9de }
kobo11-update-4.38.21908.zip: KoboRoot.tgz package { version=4.38.21908 date=2023-08-22 revision=fac580f37d610f9f83589746b785a7650532b3cf }
kobo11-update-4.39.22801.zip: KoboRoot.tgz package { version=4.39.22801 date=2024-04-02 revision=8099118e783964c8b8959136e505ee8202d8a4dc }
kobo11-update-4.39.22861.zip: KoboRoot.tgz package { version=4.39.22861 date=2024-04-24 revision=8c51b352bd251a102a10deb46c9e0c7eb5192f16 }
tolino-qt6-recovery-5.0.175773.zip: generic update.tar package { version=5.0.175773 date=2024-05-06 branch=release/v5.0.0 revision=499b0d432806417a26b4d7c44ff793839f7ba85f }
```