pgaskin/kobopatch-patches #142: Patch Increase size of kepub chapter progress chart (firmware 4.38.23171)

hatuluang, 2024-11-06:
With the release of Kobo's new firmware 4.38.23171.
I used the patch "Increase size of kepub chapter progress chart" in nickel.yaml to Kobo Clara 2E.
The base address has been changed to 0x19A150A (https://www.mobileread.com/forums/showpost.php?p=4465674&postcount=4)
I do the patch as  follows:
```
- Find:    "#chapterSizes[qApp_deviceIsDragon=true] {\n  max-height: 100px;\n  min-height: 100px;\n}"
        Replace: "#chapterSizes[qApp_deviceIsDragon=true] {\n  max-height: 170px;\n  min-height: 170px;\n  min-width: 678px;\n  max-width: 678px;\n}"
```
However, after the patch, the height of the chapter progress chart didn't change. It remained the same height size before the patch (unpatched). 
[image]

Kindly need your assistance for this issue if you have a time. 
thank you

jackiew1, 2024-11-06:
> Kindly need your assistance for this issue if you have a time. thank you

Instead of using the BaseAddress method to identify the correct CSS stream:
```  - BaseAddress: 0x19A150A```

try using this FindZlib method: 
```  - FindZlib: "#mainContainer {\n  border-bottom: 1px solid black;\n}\n#mainContainer[qApp_deviceIsTrilogy=true] {\n  qproperty-leftMargin: 20px;"```

 It works for me. The problem is that there are 2 almost identical CSS streams - one for Book chart progress and the other for Audiobook chart progress. The above FindZlib currently only matches the Book one.

hatuluang, 2024-11-06:
Unfortunately, I don't know how to use the Find Zlib method in the patch. 
It would be good if you could provide a custom patch, because It may survive better across firmware updates.

By the way, the patch works using 
`- BaseAddress: 0x19948BA`

Thanks a lot.

jackiew1, 2024-11-06:
> Unfortunately, I don't know how to use the Find Zlib method in the patch. It would be great if you could provide a custom patch or teach how to make it, as it will survive better across firmware updates.

I already provided the only info you need for the FindZlib method. 
- Delete the `BaseAddress:` line.
- Replace it with the `FindZlib:` line from my previous post. Simple copy/paste making sure you maintain the same indent.

This patch belongs to @pgaskin, not me, so I don't want to change the "official version" unless he wants me to.