pgaskin/kobopatch-patches #143: Patches for 4.41.23145

service-paradis, 2025-01-09:
I currently have the Kobo Libra Colour with firmware 4.41.23145. I was wondering if there was any plan to support it.

ragnarokxg, 2025-03-06:
I am also curious if we are going to get a patch for the new version.

ragnarokxg, 2025-03-06:
> I currently have the Kobo Libra Colour with firmware 4.41.23145. I was wondering if there was any plan to support it.

I have forked and created the patch for the new firmware. All you have to do is run the build script. 

If you do not know how I can send you the patch.

SuperSisterK, 2025-09-26:
> > I currently have the Kobo Libra Colour with firmware 4.41.23145. I was wondering if there was any plan to support it.
> 
> I have forked and created the patch for the new firmware. All you have to do is run the build script.
> 
> If you do not know how I can send you the patch.

---

ragnarok, would you mind assisting me in getting a patch for 4.43.23418? thanks kindly

ragnarokxg, 2025-09-27:
> ragnarok, would you mind assisting me in getting a patch for 4.43.23418? thanks kindly

The newest release is ready.