pgaskin/kobopatch-patches #148: Updating kobo mods for 4.39 and later

pgaskin, 2025-03-16:
For lack of a better place to put this, here's what I need to do to update my kobo stuff (mostly since 5.x was released and 4.x started branching):

```mermaid
flowchart LR
    subgraph misc
        tolino[make stuff tolino-aware?]
        ksynce[respond to emails and issue comments about kobo sync api]
    end
    subgraph kobopatch
        ku5[support 5.x update pkgs in koboutils]
        ghakpp[move kobopatch-patches to github actions]
        kpcr[review kobopatch-patches prs]
        kp5[support 5.x update pkgs in kobopatch]
        kprel4[release patches for new 4.x firmwares]
        kprel5[release patches for 5.x]
        kpbr[separate 4.x and 5.x build for patches]
        kprel[release new kobopatch version]
        kpsym[improve kobopatch symbol handling]
        ghakp[update kobopatch github actions]
        pdf5[understand multi pdf engine logic]

        subgraph rp[rewrite patches]
            p4.1[Remove PDF map widget shown during panning]
            p4.2[...]
        end

        subgraph rp5[port initial patches for 5.x]
            p5.1[...]
        end
    end
    subgraph nh[nickelhook/nickelmenu/etc]
        ntc5[new nickeltc for 5.x fw]
        nm5[new nm branch for 5.x OR rewrite nm for cross-branch support]
        nh5[update nickelhook for 5.x]
        nmsym[support complex version ranges for symbol tests]
        nmrel5[release nickelmenu for 5.x]
        nmpr[review nickelmenu prs]
    end
    subgraph dictutil
        du3[rewrite dictutil for v3 dictionaries]
        ghadu[update dictutil github actions]
        durel[release new dictutil version]
    end
    subgraph kepubify
        subgraph kpbbug[bugfixes]
            kpbbug1[bugfix: img followed by tag?]
            kpbbug2[bugfix: less strict parsing of void elements]
        end
        kpbwasm[update wasm builds]
        kpbdep[update dependences and go tc]
        kpbhtml[refactor html processing]
        ghakpb[update github actions]
        kpbrel[kepubify release]
    end
    subgraph hardware
        dev5[buy 5.x device OR port to old device OR emulate nickel]
        nboot5[update my dual-boot stuff for 5.x]
        bl5[understand recovery and bootloader for 5.x]
    end
    subgraph kobofirmware
        kfwbr[kfwproxy branching support]
        kfwp[update kobofirmware page layout]
        kfwa[update firmware release automation]
    end

    kfwbr -.-> kfwp
    kfwbr --> ghakpp
    ghakpp --> kpcr
    kpcr --> rp
    ku5 --> kp5
    ghakpp --> kp5
    rp --> kprel4
    kpbr --> kprel5
    rp -.-> rp5
    rp5 --> kprel5
    kp5 --> rp5
    kpsym --> kprel
    ghakp --> kprel
    kprel --> rp5
    ghakpp --> kpbr
    dev5 --> rp5
    nboot5 --> rp5
    kfwp --> kfwa
    du3 --> durel
    ghadu --> durel
    nh5 --> nm5
    ntc5 --> nh5
    dev5 --> nh5
    nmsym --> nmrel5
    nm5 --> nmrel5
    bl5 --> nboot5
    dev5 --> bl5
    pdf5 --> kprel5
    kpbhtml --> kpbbug
    kpbbug --> kpbrel
    kpbwasm --> kpbrel
    kpbdep --> kpbrel
    ghakpb --> kpbrel

```

pretenderlu, 2025-05-07:
That's so cool, but also so much hard work. Thank you for your dedication!

pgaskin, 2025-06-11:
And now there's also https://github.com/pgaskin/KoboStuff/issues/57.