pgaskin/kobopatch-patches #149: Issues patching 4.38.23429

jackiew1, 2025-09-09:
@pgaskin, Have you seen this thread on MobileRead https://www.mobileread.com/forums/showthread.php?t=369732 There appears to be a basic problem with patches in `libnickel.so.1.0.0.yaml` which use a `FindInstBLX SymPLT` combo, e.g. `My 10 line spacing values`, `Unify font sizes`

The odd thing is that I've had no similar problems with these same patches when patching my LibraColour with fw 4.43.23418.

pgaskin, 2025-09-10:
I'll need to look at the firmware to be sure, but I've seen this happen once before due to the compiler generating code slightly out of the range of what kobopatch recognizes.

RogerinNYC, 2025-09-10:
@pgaskin -- as a long time fan and appreciator of your work, I do dearly hope that you will find the time (and desire) in your busy schedule to take a peek and help out here. In particular, I am desperate to reclaim functionality of the Both Page Buttons Turn Next tweak (which I originally suggested to you many, many moons ago). Cheers!

jackiew1, 2025-10-06:
@pgaskin Just FYI ... 

Someone stepped forward [with a fix](https://www.mobileread.com/forums/showpost.php?p=4541243&postcount=26) for the current problem with kobopatch in v4.38.23429. He has supplied a replacement for `koboptch-windows.exe` but not equivalents for Mac/Linux.

pgaskin, 2025-10-07:
Yep, I saw that. I'll cherry-pick it into kobopatch, as this is the second time it's happened.

@alliterateaardvark